adafruit/Adafruit_Wippersnapper_Arduino #220: Support Eduroam Networks

brentru, 2022-02-23:
Add support for eduroam networks (https://eduroam.org/) used by academic institutions and some k-12.

https://github.com/martinius96/ESP32-eduroam

brentru, 2022-11-28:
https://github.com/espressif/arduino-esp32/pull/6398/files#diff-bbcb35ff3eb6ca2f03fdb22341764b9fd460885c9760cc20c24c5e55190205d8

brentru, 2023-02-01:
Forum request: https://forums.adafruit.com/viewtopic.php?p=958835#p958835

hobob0t, 2023-02-01:
Thanks for the update, I think this would be a great addition since eduroam services a very large number of students and researchers who use IoT devices.

brentru, 2023-09-06:
@hobob0t  Are you an educator? Do you have access to an Eduroam network for testing (we can gladly send hardware your way for testing)?

docmollo, 2023-09-06:
I work at a university and can easily test this, as well, if help is needed.

brentru, 2023-09-06:
@docmollo  Thank you, that would be highly useful! This is not on my _immediate_ roadmap (UART and GPS are) but it's something I can implement when taking a break from larger features.

docmollo, 2023-09-06:
You're welcome! Yeah, no worries on timing or where this in the priority...happy to help whenever something is ready to test 🤘

tyeth, 2024-11-21:
See this comment too: https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/276#issuecomment-1329760592