adafruit/Adafruit_Wippersnapper_Arduino #229: LED Toggle Inverted on HUZZAH ESP8266

brentru, 2022-03-09:
Due to how the Huzzah ESP8266 is internally wired, toggling on the WipperSnapper website is inverted.

[image]

Related Forum Issue: https://forums.adafruit.com/posting.php?mode=reply&f=66&t=189143

brentru, 2022-11-28:
@lorennorman  As this is the default behavior of the ESP8266's builtin LED - this an issue is potentially a **wontfix**.

Need your feedback:
* Would it be possible to _always_ invert the toggle on Adafruit IO for the Feather Huzzah ESP8266's builtin LED?
* Would it be better to leave the toggle alone, and document within the component configuration for this board that the inverted LED status is expected operation due to how the board is wired?

lorennorman, 2022-11-29:
Questions:
- Is this for a single board, or a whole class of boards?
- Would setting `pull: UP` fix the issue?

It seems like this is a "this component type on this board type" special case. If that's true, then yeah sorta not sure where to solve it. We certainly could add special code on the LED toggle visualization that detects that board and flips itself visually. That doesn't much worry me, unless there are a whole host of these kinds of things we expect to see over time, then I'd rather formalize it somehow.

We could add something to the JSON... I'm thinking the Board JSON for that particular pin, since that's really where the issue is. But if this is just super-rare, then I'd rather not complicate the JSON for it.

Barring any surprising answers to my questions, I'd prefer to throw a small hack into the LED toggle, with a special comment calling it out so I can find it later if these things grow.

brentru, 2022-11-29:
* Single board, Feather HUZZAH ESP8266
* No, it would not

> We certainly could add special code on the LED toggle visualization that detects that board and flips itself visually. 

Would you prefer if I handled this within the WipperSnapper firmware code, instead? I have knowledge of the device type, the LED pin, and the pin requested to toggle.

lorennorman, 2022-11-29:
Oh that's maybe the right move, especially if that were code that is `#IFDEF`'d into existence only for that board, makes it very clear that it's special case stuff. If that sounds right to you, please do!

brentru, 2022-11-29:
Sounds good to me - removed wontfix and tagged for next release.

closed 2022-12-02 by brentru