adafruit/Adafruit_Wippersnapper_Arduino #233: "Unable to connect to Adafruit IO MQTT" when following the official quick-start guide for the Arduino Nano 33 IoT

ABJ98, 2022-03-13:
**Describe the bug**
My development board can't establish a connection to Adafruit IO when I follow the official quick-start guide ([link](https://learn.adafruit.com/quickstart-adafruit-io-wippersnapper/installing-wippersnapper-on-other-hardware)) for the Arduino Nano 33 IoT, Arduino MKR WiFi 1010, and ESP8266 boards. Compiling and uploading the updated Wippersnapper_NoFS example sketch works fine on the Nano 33 IoT. Once uploaded, the board successfully establishes a connection to my home network. However, the board then fails to connect to Adafruit IO and the serial console outputs:

```
Subscribing to MQTT topics...
Running Network FSM...
Attempting to connect to WiFi...
FSM_NET_ESTABLISH_MQTT
Unable to connect to Adafruit IO MQTT, retrying in 5 seconds...
```
The device doesn't appear in the WipperSnapper dashboard. I already tried generating a new IO_KEY and used the updated value in the sketch with no success. I also updated the WiFi module firmware on the Arduino, but that didn't help either.

***Arduino board***
Arduino Nano 33 IoT

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the official setup guide
2. Install the Adafruit MQTT library
3. Upload the Wippersnapper_NoFS sketch
4. The error pops up in the serial monitor

**Expected behavior**
The board should show up on the WipperSnapper dashboard

**Which components are connected to your device**
None

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: macOS 11.6.4
 - Browser Firefox
 - Version 98

**Additional context**
Uploading to the Adafruit Feather HUZZAH ESP8266 didn't work at all, even though the guide suggests that it is a compatible device.

brentru, 2022-03-14:
I've assigned myself and will take a look at the Nano33.

> Uploading to the Adafruit Feather HUZZAH ESP8266 didn't work at all, even though the guide suggests that it is a compatible device.

Please use the WipperSnapper Firmware Uploader for the ESP8266 (https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-wippersnapper), it needs to add your credentials to the board's filesystem prior to loading the firmware.

aaron-hentschel, 2022-07-15:
I'm also experiencing the same issue as @ABJ98.  Any news on a bug fix?

brentru, 2022-07-15:
@aaron-hentschel  Are you using the Nano 33 IoT as well?

aaron-hentschel, 2022-07-15:
@brentru Yes, I am using a Nano 33 IoT.  
I'm using Adafruit Wippersnapper library version 1.0.0-beta.40 and the firmware version of the ublox module on the Nano 33 IoT is 1.4.8.

brentru, 2025-03-03:
The Arduino Nano 33 IoT is no longer supported by WipperSnapper as we are not supporting the Cortex M0 SAMD21 going forward.

closed 2025-03-03 by brentru