adafruit/Adafruit_Wippersnapper_Arduino #239: Support both Adafruit ESP32-S2 Feather Rev b/c

brentru, 2022-03-29:
> As of March 28, 2022, the ESP32-S2 Feather board has revised the power circuitry for the NeoPixel and I2C QT port. Instead of a transistor we now have a totally new LDO regulator that can be enabled or disabled with a GPIO pin. Set GPIO 7 to be output and HIGH to turn on the NeoPixel and QT power.

We should change WipperSnapper's I2C component initialization for this board (within component/i2c/...) to match the following code which detects the state of the GPIO pin and inverts it,
https://learn.adafruit.com/adafruit-esp32-s2-feather/i2c-scan-test#feather-esp32-s2-i2c-setup-3108293-1

QOXik, 2022-04-05:
In addition, they are 2 new versions of the ESP32-S2 board in revisions B and C. They have different states to turn on the power on the I2C power line. 
Rev B has to turn D7 high to enable power.
On Rev C has to turn D7 low to enable power.

brentru, 2022-04-08:
@QOXik  Thanks! Will address these revisions this upcoming week. Tagged for beta 30.

brentru, 2022-04-11:
Fixed in https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/249

closed 2022-04-11 by brentru