adafruit/Adafruit_Wippersnapper_Arduino #240: no matching function for call to 'Wippersnapper_ESP32::Wippersnapper_ESP32

Tres310, 2022-04-08:
**Describe the bug**
When running the example code, and filling in the appropriate data from 
https://learn.adafruit.com/quickstart-adafruit-io-wippersnapper/installing-wippersnapper-on-other-hardware
the code errors out: no matching function for call to 'Wippersnapper_ESP32::Wippersnapper_ESP32

***Arduino board***
Adafruit Metro ESP32-S2 Express

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://learn.adafruit.com/quickstart-adafruit-io-wippersnapper/installing-wippersnapper-on-other-hardware and run Arduino
2. Click on verify in Arduino IDE
3. See error

**Expected behavior**
WipperSnapper loaded to device.

**Which components are connected to your device**
None

**Screenshots**
See attached file. 
[image]

**Desktop (please complete the following information):**
 - OS: Win10

 - Browser FireFox/Edge
 - Version 98/100

brentru, 2022-04-08:
Are you looking to modify/add to the wippersnapper firmware, or simply install it on your hardware? Please let me know so I can properly help with this issue.

For installation, I suggest the following...

Metro ESP32-S2 Express has a precompiled UF2 file on the [releases page](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/releases/tag/1.0.0-beta.29) under the filename [wippersnapper.metroesp32s2.1.0.0-beta.29.uf2](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/releases/download/1.0.0-beta.29/wippersnapper.metroesp32s2.1.0.0-beta.29.uf2).

Please install this board's UF2 file by following the installation instructions on this guide page: https://learn.adafruit.com/quickstart-adafruit-io-wippersnapper/installing-wippersnapper-on-esp32-s2-2

PaulskPt, 2022-08-04:
Broken link: https://learn.adafruit.com/quickstart-adafruit-io-wippersnapper/installing-wippersnapper-on-esp32-s2-2

marnibrewster, 2022-09-11:
Can you post an updated link, @brentru ? Neither of the above are working and I am unable to add wippersnapper to my huzzah esp8266 via the web installer _or_ arduino ide

brentru, 2022-09-15:
@marnibrewster Could you open a new issue about the ESP8266? I can take a look there.

brentru, 2022-09-15:
The updated link would be io.adafruit.com/wippersnapper/new and selecting the ESP8266 as we've moved the installation instructions mostly off of learn, and into IO.

closed 2022-11-28 by brentru