adafruit/Adafruit_Wippersnapper_Arduino #275: WipperSnapper Removal Issues

DJDevon3, 2022-06-15:
**Describe the bug**
WipperSnapper locked to MagTag COM port, unable to flash with ESPTool or WebFlasher.

***Arduino board***
Adafruit ESP32-S2 MagTag 2.9" e-Ink display board

**To Reproduce**
Steps to reproduce the behavior:
1. Flash with TinyUF2
2. Upload WipperSnapper, board shows in USB as WIPPER
3. Reset device
4. Flash with UF2 in bootmode
5. Cannot erase flash with ESPTool or Webflasher because WIPPER boots too fast and fights to pair with COM first. Attempted to change COM port to 7 but WipperSnapper stayed connected. Couldn't release WIPPER from COM.
6. Eventually after 2 hours of trying esptool was able to beat WIPPER to the COM port, move it to COM5 on boot, and get it to erase.

**Expected behavior**
COM port open to communicate via ESPTool or Webflasher while WIPPER is USB device

**Which components are connected to your device**
Unknown, I've removed it. I had a USB device on COM1 but it didn't interfere with manually attempting to erase COM6. I did not have any other microcontrollers plugged in at the time.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Chrome Version 102.0.5005.115 (Official Build) (64-bit)
 - Python 3.10 (Windows App)
 - adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-7.3.0.uf2
 - adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-7.3.0.bin
 - tinyuf2-adafruit_magtag_29gray-0.9.4.zip
 - wippersnapper.magtag.1.0.0-beta.37.uf2
 - wippersnapper.magtag.1.0.0-beta.37.bin

**Additional context**
I tried restarting my PC to no effect multiple times. It's possible the issue stems from me flashing the bootloader back to UF2 prior to attempting to erase with ESPTool or WebFlasher. If the wrong bootloader is the culrprit then a warning in the learn guide is definitely warranted to help others avoid this mistake. For a new beginner this would be an insurmountable mistake. I spent an hour just trying to get ESPTool command to be recognized in Python.

**Python Console (Windows CMD Prompt)**

> ESP Web Flasher loaded.
> Connecting...
> Connected successfully.
> Try hard reset.
> [Object.debug:187] Finished read loop
> Error: Couldn't sync to ESP. Try resetting.

Above: WebFlasher error while WipperSnapper is already locked (paired) to COM port.

> c:\Users\Devon\Downloads>python -m esptool --chip esp32-s2 erase_flash
> esptool.py v4.1
> Found 2 serial ports
> Serial port COM6
> Connecting........................
> COM6 failed to connect: Write timeout
> Serial port COM1
> Connecting......................................
> COM1 failed to connect: Failed to connect to ESP32-S2: No serial data received.
> For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

Above: Python Error attempting to manually erase ESP32-S2 while it's locked to COM (6) port. This was attempted about 20 times to no avail trying to beat WIPPER to the COM port while resetting the board. I was stuck in a loop here for over an hour trying to get this to work.

> c:\Users\Devon\Downloads>python -m esptool --chip esp32-s2 erase_flash
> esptool.py v4.1
> Found 2 serial ports
> Serial port COM5
> Connecting...
> Chip is ESP32-S2
> Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
> Crystal is 40MHz
> MAC: 7c:df:a1:16:ca:f4
> Uploading stub...
> Running stub...
> Stub running...
> Erasing flash (this may take a while)...
> Chip erase completed successfully in 18.8s
> WARNING: ESP32-S2 chip was placed into download mode using GPIO0.
> esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
> To suppress this note, set --after option to 'no_reset'.

Above: After 2 hours of trying, I finally got ESPTool to beat WIPPER to the COM port. It changed to COM5, and erased flash successfully. 2 hours for 18.8s of glory.

> c:\Users\Devon\Downloads>python -m esptool --port COM5 --after=no_reset write_flash 0x0 adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-7.3.0.bin
> esptool.py v4.1
> Serial port COM5
> Connecting....
> Detecting chip type... Unsupported detection protocol, switching and trying again...
> Detecting chip type... ESP32-S2
> Chip is ESP32-S2
> Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
> Crystal is 40MHz
> MAC: 7c:df:a1:16:ca:f4
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Flash will be erased from 0x00000000 to 0x00167fff...
> Compressed 1471376 bytes to 958056...
> Wrote 1471376 bytes (958056 compressed) at 0x00000000 in 31.5 seconds (effective 374.0 kbit/s)...
> Hash of data verified.

Above: After flash was erased it loaded up the MagTag UF2 bin on the first attempt.

ladyada, 2022-06-17:
? pressing BOOT during power or reset should put it into ROM bootloader mode, then you  give the COM port on the command line for esptool and use `--before no_reset`

DJDevon3, 2022-06-28:
> ? pressing BOOT during power or reset should put it into ROM bootloader mode, then you give the COM port on the command line for esptool and use `--before no_reset`

Yes I booted into boot loader more than a few times while trying to get the reset button sequence right. There’s no active com port in boot loader so I just reset it again. That was part of the reset loop i got stuck in. Chicken/egg problem. Couldn’t flash the board unless I had active COM, upon reboot the COM port was instantly locked.

brentru, 2022-11-28:
@DJDevon3  Are you still having these issues? Did you resolve them?

DJDevon3, 2022-11-28:
I didnt know any better at the time @ladyada recommended the parameter. Now I know what she meant. No I didn’t try that and it sounds like that would have worked. 

Resolved by using Python’s esp tool on pc. It got caught in a race condition that eventually got it done after 50 attempts. I think it’s been fixed now but haven’t done any beta testing to confirm. Will close this issue as it’s pretty old and no one else has reported having the same issue.

closed 2022-11-28 by DJDevon3

brentru, 2022-11-28:
Thanks for following-up!