adafruit/Adafruit_Wippersnapper_Arduino #279: After install of .uf2 file, WIPPER unavailable, Neopixel shows red ( Adafruit Qt py esp32-s2)

adipotnis, 2022-07-03:
**Describe the bug**
The device named WIPPER is not available on install of the wippersnapper uf2 file.
The device neopixel shows red.

**Board**
Adafruit QT Py ESP32 -S2 with uFL

**To Reproduce**
Steps to reproduce the behavior:
1. Set device to bootloader by double pressing reset button
2. Drag and drop wippersnapper .uf2 install file
3.Shows red LED and doesn't show on finder as WIPPER

**Expected behavior**
The device should show the directory WIPPER to add the secrets.json file.

Os version: mac os 11 Big Sur

adipotnis, 2022-09-27:
I haven't used it since. I managed to set it up using the serial tool

brentru, 2022-09-27:
Ok, thanks for the update. Closing for now, will re-open if this occurs again.

closed 2022-09-27 by brentru