adafruit/Adafruit_Wippersnapper_Arduino #288: The potentiometer input utility of the ESP32-S3 is off by a factor of 3, when using wippersnapper

ron-dayney, 2022-07-29:
It appears that the ESP32-S3 is using the (internal) 11db attenuation setting in the potentiometer icon on IO. This equates to the factor of 3 error on input pin. Please change the ESP32 A/D input pin to use the 0 db setting so that I can measure true voltage. My transducer ranges from 0-2v so I do not want attenuation. The ESP32 A/D converter has 4 levels of selectable attenuation: 11, 6, 2.5 0 db. It would be very useful to be able to select the level of attenuation we want when using the potentiometer input on Adafruit IO. Is there a way you can an attenuation level selector in the potentiometer utility on IO? Thank you.

ladyada, 2022-08-01:
we would likely not make it possible to change the attenuation dynamically. it should basically be as close as possible to 0-3V as a 'standard' ARef. whats it now?

ron-dayney, 2022-08-01:
Currently, it appears that it is set at the default of 11 dB level. This is why my readings are off by a factor of 3. It does allow for voltage input of 0-10v now. The closest setting to 0-3v would be 0 dB level. That would give it a range of 0-3.3V full scale. I think that would be better since most sensors are geared toward the lower voltages of 3.3 V logic. However you probably know a lot more about the voltage level of most sensors, so you decide the "best" attenuation level, if we can only have one level. I understand making the attenuation level dynamically selectable may be too tricky, Just trying to make suggestions for future features that would make Adafruit IO more desirable. Since we are in Beta. How about making an extra potentiometer icon/utility that is fixed at 0 dB? Perhaps the labels could be changed to voltage levels like, 0-10 V analog in, and for the new one, 0-3 V analog in. The analog in feature might take separate icons for each microcontroller since the Huzza A/D is only 0-1 V, and it has no built in attenuator. The built in attenuator feature in the ESP32 is actually a very cool feature. This is not urgent for me, just trying to be a contributor. The Adafruit IO is the coolest thing I have seen in this growing IOT industry.  Keep up the truly amazing and awesome work there. Thank you.

ladyada, 2022-08-01:
ok great info - @brentru will check when they are back to work this week :)