adafruit/Adafruit_Wippersnapper_Arduino #290: ESP32-S3 feather and Huzza 8266 feather go offline and need resetting very often

ron-dayney, 2022-08-02:
The Feather ESP32-S3 goes offline often. Pressing reset normally re-establishes communication. This happens regardless of the sample rate of a QWICC sensor (I've tried 15 min down to 30 sec.) If I sit and monitor the Devices screen, I can see it go offline and come back online very often. Eventually it stops re-establishing connection, and I have to reset.

***Arduino board***
Feather ESP32-S3

**To Reproduce**
Steps to reproduce the behavior:
1. Use the above processor, the ESP32-S3
2. Connect QWICC sensors
3. Create wipersnapper device and download to device, create feeds, create dashboards, assign feeds to dashboards
4. monitor device LEDS on ESP32-S3 and devices screen on IO to observe it going offline 

**Expected behavior**
At random, frequent intervals the ESP32-S3 goes offline, until I hit the reset button.

**Which components are connected to your device**
ESP32- QWICC port-AHT-20 temp&humidity; A8 to wipper of 10K potentiometer hooked from 3v to gnd

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 103

**Additional context**
My ISP is Skyfiber. It is a wireless provider. I am in a rural area just outside Reno, NV. I get 15 Mbps down and 10 Mbps up. My idle latency is 47 mS, upload latency is 127mS, download latency is 107 mS (as reported by Speedtest.net).
I am using a Netgear Orbi Mesh router as the DHCP server and gateway.
The device is powered by a USB hub with 4 A. switching power supply. It and the Huzza 8266 feather that I also have on Adafruit IO are the only devices powered by that hub. The feather Huzza 8266 goes offline once in a while too, but not nearly as much as the ESP32-S3.

brentru, 2022-08-04:
@ron-dayney  Do you have any idea of how frequently this occurs? Does your Feather S3 disconnect when not connected to sensors?

ron-dayney, 2022-08-08:
It was occurring every 2-3 hours with Qwiic sensor attached. Per your suggestion, I removed the sensors, and now it has to be reset around every 8-10 hours. The Feather Huzza 8266 is more reliable, but still needs to be reset once every 24 hours or so. The Metro ESP32 S2 has been the most reliable, so far have never needed to reset it. I just setup a 4th device a Feather ESP32 S2 and it has been reliable too. Looks like the disconnecting problem may be with just the S3 version. Darn, I just bought two more S3's hoping they would be reliable on Adafruit IO. I have not tried the new S3's yet.

brentru, 2022-08-08:
> Looks like the disconnecting problem may be with just the S3 version. Darn, I just bought two more S3's hoping they would be reliable on Adafruit IO. I have not tried the new S3's yet.

Are you able to tell if the disconnection is due to your internet dropping out (you can check this with a serial monitor open when running wippersnapper)? The WipperSnapper firmware does perform a watchdog timer reset when it's unable to send data to Adafruit IO, or loses connection. It shouldn't go fully offline, though.

I'm going to set up a S3 feather here to test.

ron-dayney, 2022-08-08:
I thought it might be my ISP connection too, until I has a couple ESP32 S2's running reliably sitting on the same connection and power supply. I am still testing too, but the Metro ESP S2 has not gone offline in several days. The Feather S2 has been steady since I put it on 12 hours ago. If it were my ISP connection, then all devices would have gone offline similarly. Still, I would like to understand what you mean by monitoring a wippersnapper device with serial monitor. I am running Win 10, is there a serial terminal program I need to do this? Do I plug in the wippersnapper to the USB port on my Win 10 machine and run terminal emulator? I also have a Rasberry Pi 4, or a Ubuntu Laptop but am not proficient with Linux. If you can send link to instructions I would be very grateful. I do have a couple (new) Feather S3's with TFT display built in. Can that be used as a serial terminal?

brentru, 2022-08-12:
> Still, I would like to understand what you mean by monitoring a wippersnapper device with serial monitor.

WipperSnapper is an Arduino application. You should be able to plug in a board running wippersnapper and open Arduino's serial monitor (we have a guide here on usage - https://learn.adafruit.com/adafruit-arduino-lesson-5-the-serial-monitor) to obtain the debug logs from the device.

We're thinking about doing a web-based version of this (can open the device's debug output from Adafruit IO) but that's a bit in the future.

brentru, 2022-08-12:
I've set up a Feather ESP32-S3 and a dashboard to let this soak test over the weekend. I'll let you know the results!

[image]

brentru, 2022-08-15:
@ron-dayney  Ok, I left my Feather S3 online all weekend. I'm unable to find any gaps in the data, nor did it go totally offline requiring a restart. 
[image]

ron-dayney, 2022-08-30:
After your feedback, and after I updated all my devices to the latest version of Adafruit IO, I decided to try one of the new ESP32-S3 feather that I purchased recently. It works great and has been continuously online for several days now without a single hang up. I have determined that the other S3 is bad. Per your advice I even hooked up that S3 to a serial port to "observe" it. It has progressively gotten worse from the day I installed it. Now it locks up in under 2 hours. It started out locking up every 12 hours or so. Most of the time it hangs the serial port too and all text to the serial monitor just stop. There was one time the serial monitor seemed to catch an error right before a hang up "Client sendPacket returned: 111", then "unable to connect to Adafruit IO MQTT". Since a new S3 works fine, it appears to be a bad hardware issue with that particular one. Please remove the bad S3 that is offline in my IO account. I have no way of removing it myself. Is there any warranty on the bad S3? I purchased it a few months back. Should I send it to you?