adafruit/Adafruit_Wippersnapper_Arduino #292: wippersnapper_demo - please indicate that user has to install 10 librares (name them)

PaulskPt, 2022-08-02:
In the first place thank you for creating a demo to build inside the Arduino IDE 
and trying to make Wippersnapper known to the community!

Today I tried to build the 'wippersnapper_demo' onto an Adafruit MAGTAG. I cost me at least an hour to ultimatly get an error free build.
Ten times I was confronted with 'file not found' errors. 
Each error - and Google'ing for Adafruit products with the same name - pointed me to a library I had to install. 
Inside the Arduino IDE v1.8.19 this cost time: one has to go to > Tools > Manage libraries ... then wait until the IDE is ready to accept your search string.... install the library... exit ... wait until the IDE is ready to take your commands, in my case: try to build again.

If the README.md of the Adafruit_Wippersnapper_Arduino repo (or the demo sketch itself) had listed the ten libraries it needed to build successfully, it would have been much less complicate and cost me much less time: do one session of installing ten libraries.
```
The list of (not mentioned) libraries I needed to install are:
- Adafruit_Dotstar;
- Adafruit_TinyUSB;
- Adafruit_BME280;
- Adafruit_DPS310;
- Adafruit_MCP9600 (for the MCP9601);
- Adafruit_MCP9808;
- Adafruit_SCD30
- Sensirion I2C SCD4x;
- TLs2591;
- Adafruit_MQTT.
```
Please update the README.md or put it elsewhere the user can take notice of this (e.g.: put it in the demo sktech itself).

By the way: the demo is quite 'meager'. The output in the Monitor window only shows a 'litany' of 'PING!' lines.
It would have been nice to see all the MQTT packets. From that one learns. Not from 'PING!' only.

PaulskPt, 2022-08-02:
Update: I just tried PuTTY. The image below shows some more (boot) info:

[image]

ladyada, 2022-08-02:
the library has a dependancy list
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/main/library.properties
if you isntalled the library from the library manager in the IDE, it will install all of them for you - the list is going to change constantly so we would not maintain a separate list

PaulskPt, 2022-08-02:
Thank you for the rapid response! Yep, I downloaded the repo and moved it to the documents/arduino/libraries folder. So, I installed wrong. Mea culpa! Issue closed

closed 2022-08-02 by PaulskPt