adafruit/Adafruit_Wippersnapper_Arduino #299: MCP9601 doens't work on ESP32-V2 Feather

evaherrada, 2022-08-22:
**Describe the bug**
On the ESP32-V2 Feather, the MCP9601 shows up for a few seconds after initially being plugged in. If I do an i2c scan immediately I can see it, but when I click scan again it disappears.I tested on Arduino and it wouldn't show up at all. I tested on CircuitPython with a 

***Arduino board***
INSERT ARDUINO BOARD NAME/TYPE HERE

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

brentru, 2022-11-28:
@lorennorman  - The MCP9601 is finicky over I2C and may not be appropriate for WipperSnapper due to it locking up after the MCU performs the (required) initial I2C bus scan.

Would it be possible for you to remove it from the picker? I can remove it from the core.

lorennorman, 2022-11-29:
@brentru would you like to set `published: false` in the component JSON so it switches to "in-development", or should we remove it altogether? (I'll run some reports on prod to make sure it isn't in widespread use with a lot of history.)

brentru, 2022-11-29:
@lorennorman  I'd like to fully remove it.

brentru, 2022-11-29:
Let me know when done pls!

lorennorman, 2022-11-29:
Done!

brentru, 2022-11-30:
Thanks, the MCP9601 will be fully removed from the firmware in the next release.

rob-vaughn, 2023-03-24:
@brentru are there any plans to add a thermocouple amplifier like the mcp9601 or will they not work with WhipperSnapper?

brentru, 2023-03-27:
@rob-vaughn  The MCP960x series does not work properly with WipperSnapper because the chip "locks" after an I2C scan operation. "under the hood", wippersnapper always sends an i2c scan command to the board to check if an i2c device exists, then initializes the sensor. This combination of commands causes the MCP960x series to lock up and not function properly. 

I believe we support the MAX31856 and the DS18B20 digital temperature sensors though.

brentru, 2023-03-27:
> Thanks Brent!
>
> Any plans to add a thermistor like the NTC 3950? These are similar to
> thermocouples in size, but more accurate.

We do sell this thermistor - https://www.adafruit.com/product/372.. What's
the advantage of this vs a DS18b20 for you?

rob-vaughn, 2023-03-27:
I guess they are similar. Thermistor sensor heads might be a little
smaller. It's for a specific application in a sap flow sensor that needs to
contact the surface of a stem to measure temperature change. I might try
the DS18b20 and see how it works.

Thanks!

brentru, 2023-03-27:
Try the DS18b20, the temperature readings are more precise, and let me know
how it goes. We can work on adding the NTC 3950 via a "generic thermistor"
component if you'd like, too!