adafruit/Adafruit_Wippersnapper_Arduino #308: WipperSnapper loaded on Qt Py ESP32-C3 goes into infinite boot loop

sanramaraj, 2022-08-31:
**Describe the bug**
WipperSnapper loaded to Qt Py ESP32-C3 from Adafruit.io website (below link) was successful but the device goes into a continuous boot loop evident from continuous connect/disconnect sound from the computer.

https://adafruit.github.io/WipperSnapper_Firmware_Uploader/?board=qtpy-esp32c3

Console log attached.

***Arduino board***
Qt Py ESP32-C3

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Adafruit.io
2. Click on 'Add Device'
3. Scroll down and select 'Qt Py ESP32-C3'
4. Click on 'Get Firmware'
5. Click on 'Connect'
6. Select the port with keywords - USB JTAG/Serial
7. Enter the credentials for the wifi network and Adafruit IO
8. Click on 'Install WipperSnapper'
9. Wait for the text 'To run the new firmware, please reset your device.' to appear in the console log
10.  Reset the device
11.  Leave the device connected to the computer
12.  You'll hear continuous connect/disconnect sounds and the LED on board will never turn-on

**Expected behavior**
After installing WipperSnapper successfully, expect the LED to show some indication that it booted up properly and observe the ESP32-C3 on the Devices page of Adafruit.io

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?
None

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win10 Prof, Win11 Home
 - Browser: Chrome
 - Version: xx

**Additional context**
Add any other context about the problem here.

[QtPy_C3_Log.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/9457620/QtPy_C3_Log.txt)

sanramaraj, 2022-08-31:
@marnibrewster do you see the LED light up when you added the guest network for 2.4GHz? if left connected to the computer, does it continuously connect/disconnect?

brentru, 2022-08-31:
@marnibrewster @sanramaraj  Are you both having this issue with Qt Py ESP32-C3? Did it occur on previous Wippersnapper releases, or just the latest?

sanramaraj, 2022-08-31:
@brentru I started using it a month ago, I was never successful with WipperSnapper installation from the website. The most recent attempt was 2 days ago.

marnibrewster, 2022-08-31:
I just received my c3 a day ago, so this is my first attempt with it.
I can upload code via Arduino just fine. I hit snags when trying to upload circuit python code though. Can provide more details after work today. 
@sanramaraj The LEDs stopped working when I first attempted to install wippersnapper.

marnibrewster, 2022-08-31:
@brentru are there manual install instructions somewhere?

sanramaraj, 2022-08-31:
@marnibrewster I had the same experience, LED stopped working after the first attempt to install WipperSnapper

marnibrewster, 2022-08-31:
I'm getting an error when trying to install the `urequests` package in Thonny now: 

> "This doesn't look like MicroPython/CircuitPython package. Are you sure you want to install it?"

marnibrewster, 2022-08-31:
I just received my second c3 and attempted the wippersnapper install on it. LEDs working when I initially plugged it in. Held boot, pressed reset, let go of boot. Was able to find the port on the wippersnapper install page. Installation seemingly succeeded and I was told to reset. The LED was green! I reset the c3 with the reset button. Nothing happened (device didn't show up in my device list). I tried resetting by unplugging and replugging. Nothing. 
It was using: `wippersnapper.qtpy_esp32c3.littlefs.1.0.0-beta.45.combined.bin`
Quite frustrating!

marnibrewster, 2022-08-31:
@brentru I can try re-installing wippersnapper on my feather huzzah esp8266 tonight and will get back to you

sanramaraj, 2022-08-31:
@marnibrewster I guess there is something messed up with the WipperSnapper for ESP32-C3

marnibrewster, 2022-09-01:
@sanramaraj I should add that while I didn't hear the noises you heard, I did notice something weird after a seemingly successful install of wippersnapper. If I reset the board, go back to the wippersnapper install page, and click 'connect', the c3's port flashes on and then off repeatedly.

brentru, 2022-09-02:
@marnibrewster  @sanramaraj  Ok, thank you both for the info. This was previously an issue preventing us from adding C3 in the first place, it has been fixed in the latest BSP (https://github.com/espressif/arduino-esp32/issues/6862#issuecomment-1175926313) and there are [similar issues](https://github.com/tzapu/WiFiManager/issues/1482) in other libraries. 

@marnibrewster  There are manual installation instructions on this page: https://learn.adafruit.com/how-to-add-a-new-board-to-wippersnapper/build-wippersnapper

I've added this to the top of my stack for next week and will look at it after the USA holiday.

marnibrewster, 2022-09-03:
@brentru Thank you! 

I tried the wippersnapper manual install and was hitting littlefs issues 🤔 
I was able to get my project working (using the qt py with a scd4x temp/humidity/co2 sensor, pushing data to adafruit io, and then using the data in homebridge to make it available on my homekit) using adafruitio mqtt and the sensirion library for now. 

Let me know if there's anything I can contribute / test next week.

marnibrewster, 2022-09-11:
@brentru I was unable to update my huzzah esp8266 to the latest beta (same experience: the upload succeeded, I was told to reset my device, but the same version was on it and it wouldn't connect to IO). After removing it as a device in IO, I can no longer upload wippersnapper via the web tool.

brentru, 2022-09-14:
@sanramaraj No estimate, it's the next item on my list after https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/321 (and associated refactoring around i2c, updating documentation) is done today.

brentru, 2022-09-15:
@marnibrewster @sanramaraj  This issue should be resolved as of #326. We patched [a line of code causing ESPTool to merge the binary for the ESP32 platform as a target, instead of the ESP32C3 platform](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/actions/runs/3061934377/jobs/4942394014#step:16:1). This should be resolved going forwards.

I tested this with my QT Py ESP32C3 and did not get the boot loop when using the web uploader.

Please re-open this issue if you are still experiencing the C3 boot-looping.

closed 2022-09-15 by brentru

sanramaraj, 2022-09-15:
It works!!

marnibrewster, 2022-09-15:
I can't wait to try it out tonight, thank you so much for the fast resolution on this!