adafruit/Adafruit_Wippersnapper_Arduino #313: Servo Componet not adding on QTPY 32-S2

djecken, 2022-09-07:
**Describe the bug**
servo component does not load on qtpy 32 s2. Appears to add if the device is offline but doesn’t run the servo

***Arduino board***
QTPY 32 S2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'add component'
2. Click on 'servo'
3. Component element loads then disappears  

**Expected behavior**
loads and works on QTPY 32

**Which components are connected to your device**
pir + servo

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser chrome
 - Version 104.0.5112.101

brentru, 2022-09-07:
also reported by @caternuson

brentru, 2022-09-12:
@djecken @caternuson This bug has been addressed in #320  and merged and released in beta 49 (https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/releases/tag/1.0.0-beta.48).

Please update your firmware to the absolute latest. On Adafruit IO, delete the servo component from your ESP32-S2 board. Then, add it back. 

Please note that Pin A0 does not work properly with the ESP32-S2 QT Py, however, A1 through A3 do work. This will be noted in a guide later.