adafruit/Adafruit_Wippersnapper_Arduino #315: Servo component stops working after power cycle

caternuson, 2022-09-08:
**Describe the bug**
Servo stops updating after a power cycle. Can get it working again by either (1) clicking cog icon and then clicking "Update Component" without making any actual changes, or (2) deleting and recreating a new servo component.

***Arduino board***
Board is a QT PY ESP32 Pico. Servo is a standard micro hobby servo (SG92R) connected to MOSI. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add a servo component. Use slider to confirm it's working. Servo should move as expected.
2. Power cycle setup.
3. Once board shows as Online again, try moving servo slider.
4. Servo does not move.

**Expected behavior**
Servo should move again once board shows as Online.

**Which components are connected to your device**
Servo

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser: Firefox
 - Version: 104.0.1

**Smartphone (please complete the following information):**
None

**Additional context**
None

caternuson, 2022-09-09:
**Update** - this issue may be more general. I'm seeing the same behavior with a Push Button component.

brentru, 2022-09-09:
cc @lorennorman - are we still sending the hardware re-sync packets?

brentru, 2022-09-15:
@caternuson  Marking this as resolved, we deployed a fix for this to production and it was tested on hardware. Please re-open if you're still having this issue.

closed 2022-09-15 by brentru

lorennorman, 2022-09-16:
followup, i believe i fixed this with the servo slider primarily. i'm now realizing other components might be affected and am checking them separately.

thank you @caternuson!

lorennorman, 2022-10-11:
Just saw this, totally forgot about that portion of the code …I think. I’m
guessing we didn’t think to test this!

Will fix first thing Monday and send it out with the update PR. I’ll also
do… something to help remember all the update spots for each component
type.