adafruit/Adafruit_Wippersnapper_Arduino #323: Updating component doesn't respect current state - Relay

tyeth, 2022-09-13:
**Describe the bug**
Add a relay (on mine off is on), switch relay to On, i.e. Not Default toggle option, then goto component and update label or icon for one of the toggle states. Relay gets toggled back to default after saving update to component info, but toggle button remains in opposite state. Double toggling the toggle button returns page to in sync with relay. Updating the component again recreates the same behaviour.

***Arduino board***
QTPY ESP32-S2

**Which components are connected to your device**
SCD40, Si7021, Relay x 2 (on MOSI and MISO pins)

brentru, 2022-09-14:
@lorennorman  This seems like an issue during re-synchronization to me, unless you think it's a firmware issue?

tyeth, 2022-09-14:
I think you're right, although I don't yet have full confidence in my firmware until someone else tests it too, but it does appear to work... 🤞

lorennorman, 2022-09-15:
My gut tells me that this an issue with my web code, I'll try to reproduce and see what I discover.

Thanks for the ongoing contributions and the excellent bug report @tyeth!

lorennorman, 2022-09-22:
I'm unable to reproduce any weird behavior here so far, but I do not have the hardware in question to test against, and as I keep rereading the original description I'm thinking that is the only place we see the bug.

@tyeth are you saying that the web interface remains consistent, but the hardware itself gets toggled upon an update, even if that update is merely cosmetic (label/icon changes)?

I'm also not sure I understand when you say "on mine off is on", is that normal/expected, or also possible indication of a bug?

@brentru do you have this hardware to test against? have you reproduced this bug if so?

lorennorman, 2022-09-22:
Wait nevermind I think I'm onto this now, will report back soon...

lorennorman, 2022-09-22:
Figured out and fixed this one!

@tyeth Please refresh and try this again (update one of your relays by only changing a label or icon), your device will not experience a desync upon such updates anymore (physical relay will not lose its value during the update.)

You uncovered a deeper issue with how we update components, as well. _(Thank you!)_ I'm going to close this issue and open a fresh one specific to the deeper problem. It will take us a bit longer to work up a solution as it deals with protobuf updates and is an opportunity for some performance/bandwidth/size improvements, so we want to get it right.

closed 2022-09-22 by lorennorman