adafruit/Adafruit_Wippersnapper_Arduino #328: Scan SSID/RSSI before connecting to WiFi

brentru, 2022-09-15:
To help debug users who are not sure if they're connecting their WipperSnapper boards to a wireless network unsupported by their radio (5GHz vs 2.4GHz required for ESPx), we may want to perform a full SSID scan.

This would occur before we attempt to connect to WiFi. It would also log (to the .txt boot file)/print the RSSI values for each wireless network to assist engineering support with debug logs.

Additionally, WipperSnapper may want to raise/log an error and halt if the WiFi network the user is attempting to connect to is not found by the preliminary scan (if the hardware can't detect it, we shouldn't attempt to connect to it).

tyeth, 2022-09-20:
I found the debug serial info for wifi came too quickly for me to receive it 29times out of 30, adding things to boot txt makes sense in erroneous states like broken/missing secrets or network not found (so list available+rssi).
  I'd love to also include a second network or ideally a few (4) or array. Use case is programming at the pc on home wifi then using the device on another network in-situ. I dream of wpa_supplicant.conf compatibility for copy-pasta ;)
  I know eventually you wish to also support GSM/LTE modem connections, and I had a quick gander at the different connect routines while debugging, so I wonder how the connect routine could be improved to do a scan, and select from one of the known networks with the best RSSI. That's what I ended up doing on micropython whenever it disconnected or started because it was easy-ish compared to the lower level esp-idf...

brentru, 2022-09-21:
> I found the debug serial info for wifi came too quickly for me to receive it 29times out of 30, adding things to boot txt makes sense in erroneous states like broken/missing secrets or network not found (so list available+rssi).

It sometimes does flake at the serial, but, we can have it printed out multiple times. Some boards, any flashed with LittleFS/Web Uploader, do not support the boot txt file so we'll need both methods here.

> I'd love to also include a second network or ideally a few (4) or array. Use case is programming at the pc on home wifi then using the device on another network in-situ

Totally possible and useful for education environments. May be a separate PR, though.

>  dream of wpa_supplicant.conf compatibility for copy-pasta ;)

We'll follow CircuitPython's lead with their WiFi configuration file

>  so I wonder how the connect routine could be improved to do a scan, and select from one of the known networks with the best RSSI.

Ideally, the user sets up their network, the software scans, and then attempts to connect to it if found. If multiple networks are configured, as mentioned above, possibly we'll fall back if the first one isn't found. That'll be a separate PR/Issue, though.

tyeth, 2022-09-21:
> It sometimes does flake at the serial

I seemed to get myself confused into thinking I was missing a debug flag due to never seeing anything before ```Attempting to connect to wifi```. I reset the board each time, wait for explorer to load the file browser for the usb and then open serial monitor, maybe 1-1.5seconds after hitting reset button in total, and 0.5-1s max after usb reconnects. Only once I have seen the full serial debug printing device name and wifi details. Here is an example 
[image]

> We'll follow CircuitPython's lead with their WiFi configuration file

Yeah that makes total sense. I wonder where I saw the equivalent of ```CIRCUITPY_WIFI_SSID2=```, must have been some micropython example instead.

 I think long term I'd love the support of single object and/or array for networks, and to potentially add a key for priority, or potentially just use RSSI based priority upon start/disconnect. i.e. currently it's: 
```json
"network_type_wifi":{
		"network_ssid":"free4all_2G",
		"network_password":"password"
	}
```
but I'd like to support that (backwards compatible) and also:
```json
"network_type_wifi":[
          {
		"network_ssid":"my_house",
		"network_password":"password"
	},
        {
		"network_ssid":"the_garden",
		"network_password":"password"
	}
]
```

Definitely agree they are separate pull requests/features to this. Let's say thoughts and extensions to the idea of providing scan feedback if network (re)connect fails.

brentru, 2022-11-29:
@tyeth  

> It is possible to leave the serial monitor open if the port number is unchanged, but it takes too long to wake-up and usually misses the first messages. I'

I'd change the arduino code to add `  while (!Serial) delay(10);`:

```
void setup() {
  // Provisioning must occur prior to serial init.
  wipper.provision();

  Serial.begin(115200);
  while (!Serial) delay(10);

  wipper.connect();

}
```

closed 2022-11-30 by brentru