adafruit/Adafruit_Wippersnapper_Arduino #330: upgraded to beta 50 on v2 crashes right away

TheGodMotherG, 2022-09-22:
**Describe the bug**
I had version 44 working perfectly,, i dont know why i decided to mess with it. now when i install beta 50(with the web flasher) non of the lights turn on it does not even recognize a new device in the io and i get this message in serial monitor.

Tried 3 computers mac and windows and 4 different wires

This sensor controls something very important in my house i really cant go without it please help

***Arduino board***

https://www.adafruit.com/product/5400
Adafruit ESP32 Feather V2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'updater'
2. Click on 'update'
3.  completes with no error
4. See error

**Expected behavior**
I expected it to work like it did in version 44 beta when tried to upgrade it does not work the lights dont turn on 

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?

**Screenshots**
22:36:59.171 -> ELF file SHA256: 0000000000000000
22:36:59.171 -> 
22:36:59.171 -> Rebooting...
22:36:59.171 -> ets Jul 29 2019 12:21:46
22:36:59.219 -> 
22:36:59.219 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:36:59.219 -> configsip: 271414342, SPIWP:0xee
22:36:59.219 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:36:59.219 -> mode:DIO, clock div:1
22:36:59.219 -> load:0x3fff0030,len:1184
22:36:59.219 -> load:0x40078000,len:13160
22:36:59.219 -> load:0x40080400,len:3036
22:36:59.219 -> entry 0x400805e4
22:36:59.778 -> 
22:36:59.778 -> abort() was called at PC 0x4008690b on core 1
22:36:59.778 -> 
22:36:59.778 -> 
22:36:59.778 -> Backtrace:0x40084281:0x3ffb25f00x4008d659:0x3ffb2610 0x40092ae1:0x3ffb2630 0x4008690b:0x3ffb26b0 0x40086a55:0x3ffb26e0 0x400e8d74:0x3ffb2700 0x400dad09:0x3ffb2740 0x400dabc9:0x3ffb2790 0x400d7c2b:0x3ffb27b0 0x400d3630:0x3ffb27d0 0x400d2ec9:0x3ffb27f0 0x400e4ed6:0x3ffb2820 
22:36:59.826 -> 
22:36:59.826 -> 
22:36:59.826 -> 
22:36:59.826 -> 
22:36:59.826 -> ELF file SHA256: 0000000000000000
22:36:59.826 -> 
22:36:59.826 -> Rebooting...
22:36:59.826 -> ets Jul 29 2019 12:21:46
22:36:59.826 -> 
22:36:59.826 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:36:59.826 -> configsip: 271414342, SPIWP:0xee
22:36:59.826 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:36:59.826 -> mode:DIO, clock div:1
22:36:59.873 -> load:0x3fff0030,len:1184
22:36:59.873 -> load:0x40078000,len:13160
22:36:59.873 -> load:0x40080400,len:3036
22:36:59.873 -> entry 0x400805e4
22:37:00.433 -> 
22:37:00.433 -> abort() was called at PC 0x4008690b on core 1
22:37:00.433 -> 
22:37:00.433 -> 
22:37:00.433 -> Backtrace:0x40084281:0x3ffb25f00x4008d659:0x3ffb2610 0x40092ae1:0x3ffb2630 0x4008690b:0x3ffb26b0 0x40086a55:0x3ffb26e0 0x400e8d74:0x3ffb2700 0x400dad09:0x3ffb2740 0x400dabc9:0x3ffb2790 0x400d7c2b:0x3ffb27b0 0x400d3630:0x3ffb27d0 0x400d2ec9:0x3ffb27f0 0x400e4ed6:0x3ffb2820 
22:37:00.480 -> 
22:37:00.480 -> 
22:37:00.480 -> 
22:37:00.480 -> 
22:37:00.480 -> ELF file SHA256: 0000000000000000
22:37:00.480 -> 
22:37:00.480 -> Rebooting...
22:37:00.480 -> ets Jul 29 2019 12:21:46
22:37:00.480 -> 
22:37:00.480 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:37:00.480 -> configsip: 271414342, SPIWP:0xee
22:37:00.480 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:37:00.480 -> mode:DIO, clock div:1
22:37:00.480 -> load:0x3fff0030,len:1184
22:37:00.480 -> load:0x40078000,len:13160
22:37:00.480 -> load:0x40080400,len:3036
22:37:00.480 -> entry 0x400805e4

**Desktop (please complete the following information):**
 - Tried different operating systems and browsers

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
if i can somehow go back to the old version while i wait for a fix that would be great

brentru, 2022-09-22:
Also reported by @tyeth  here

https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/325#issuecomment-1255074926

brentru, 2022-09-22:
We've fixed the bug with the updater - you should be able to downgrade to Beta 49 at this point using the "New Device" button on io.adafruit.com or by visiting the web uploader.

I'll update this post when we've re-released a working version of Beta 50.

tyeth, 2022-09-22:
@TheGodMotherG I've been saved in the past by using previous releases on a repository by going to the /releases page: 
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/releases
Sorry if I broke it. On a related note @brentru I have just installed arduino ide 2.0 and there was a newer board support package for esp32, v2.0.5 rather than 2.0.4. Could the build have one step/task using a pinned version and another using latest and somehow that blew us up?

brentru, 2022-09-22:
you did not break it. there's an issue with adafruit neopixel and the latest Arduino-ESP32 core  (which beta.50 was built on) 

https://github.com/adafruit/Adafruit_NeoPixel/pull/330

tyeth, 2022-09-22:
Strangely I do get a problem if I use ESP32 board support package 2.0.5. It gives me a purple flashing light and no drive/serial. If I compile and upload with 2.0.4 it uploads and works as expected. v50 compile from Wippersnapper_Demo.ino not via UF2

brentru, 2022-09-22:
Yup, we did a release of Adafruit_NeoPixel with the fix and it should be picked up by the library manager in a few hours. We'll re-release WipperSnapper Beta 50 when it's all set.

brentru, 2022-09-22:
WipperSnapper Beta 50 has been re-released with a patch to fix the crashing issue under Adafruit_NeoPixel. 

If the same issue persists in Beta 50, or if are experiencing a new issue - please file a new bug report on https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/new

closed 2022-09-22 by brentru

TheGodMotherG, 2022-09-23:
Still does not work with fixed beta50 on esp32 v2 just get a red light. and when i downgrade to version 49 and try to use the sht40 sensor it says firmware out of date cannot use this component. i downgraded every version from 49-43 still same issue not letting it work with sht40. very strange because i had it working on version 44 i think just yesterday before attempting to upgrade