adafruit/Adafruit_Wippersnapper_Arduino #333: An edited component loses its value on the device

lorennorman, 2022-09-22:
**Describe the bug**
Existing output pin components lose their value when their settings are updated.

***Arduino board***
All boards, this is a system issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect and set up a WipperSnapper device with an LED (stay connected for the duration)
4. Add an LED Component via the web, but use the wrong Pin for it
5. Toggle the LED "on" in the web interface
6. Edit the LED component, set the correct pin this time, and submit
7. Observe the device: the LED does not come on

**Expected behavior**
The LED should get the current value of the feed it is connected to (in this case a value of '1', indicating it should be on.)

**Which components are connected to your device**
This generalizes to all pin components with a direction of "OUTPUT".

**Additional context**
This was discovered by @tyeth and reported in #323

brentru, 2022-11-28:
@tyeth and @lorennorman  - Is this issue still occurring?

brentru, 2022-11-29:
@lorennorman  ok - labeling as a broker fix, feel free to transfer the issue.