adafruit/Adafruit_Wippersnapper_Arduino #334: Same issue as I reported.

crustylarry, 2022-09-22:
FSM_NET_ESTABLISH_MQTT

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40222276 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  40106255 00497541 3fff1ffc 00000000
3ffffeb0:  3fff0a40 3fff1ffc 0000012c 4020d98e
3ffffec0:  000000ff 3fff1ffc 3fff2eec 4020d8ae
3ffffed0:  4021731d 0000012c 00000014 40217312
3ffffee0:  00000005 0000012c 00000000 00000014
3ffffef0:  00000005 3fff18ac 3fff0ea8 40208581
3fffff00:  40214dc8 3ffeb180 3fff1c14 40215034
3fffff10:  00000001 00000016 3fff1c14 40215118
3fffff20:  000000a3 00000090 3fff19ac 00000000
3fffff30:  3fff18ac 3fff0ea8 3fff1c14 402088db
3fffff40:  00000037 00000027 ffffffff 00000001
3fffff50:  433a3843 33413a39 3a30393a 323a3733
3fffff60:  00000037 00000001 3fff1c14 40214e74
3fffff70:  00000000 00000000 3fff0ea8 3fff1ca0
3fffff80:  0001c200 0000001c 00000000 3fff1ca0
3fffff90:  3fffdad0 00000000 3fff0ea8 40207189
3fffffa0:  feefeffe feefeffe 3fff1c8c 40215f4c
3fffffb0:  feefeffe feefeffe 3ffe8600 401013f9
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00060200
~ld
Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.50
Board ID: feather-esp8266
Adafruit.io User: K9LD
WiFi Network: Ma Pa Dalton 2.4GHz
MAC Address: C8:C9:A3:90:37:27
-------------------------------
Subscribing to MQTT topics...
Running Network FSM...
Attempting to connect to WiFi...
CONNECTING
FSM_NET_ESTABLISH_MQTT

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40222276 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  40106255 00402121 3fff1ffc 00000000
3ffffeb0:  3fff0a40 3fff1ffc 0000012c 4020d98e
3ffffec0:  000000ff 3fff1ffc 3fff2eec 4020d8ae
3ffffed0:  4021731d 0000012c 00000014 40217312
3ffffee0:  00000005 0000012c 00000000 00000014
3ffffef0:  00000005 3fff18ac 3fff0ea8 40208581
3fffff00:  40214dc8 3ffeb180 3fff1c14 40215034
3fffff10:  00000001 00000016 3fff1c14 40215118
3fffff20:  000000a3 00000090 3fff19ac 00000000
3fffff30:  3fff18ac 3fff0ea8 3fff1c14 402088db
3fffff40:  00000037 00000027 ffffffff 00000001
3fffff50:  433a3843 33413a39 3a30393a 323a3733
3fffff60:  00000037 00000001 3fff1c14 40214e74
3fffff70:  00000000 00000000 3fff0ea8 3fff1ca0
3fffff80:  0001c200 0000001c 00000000 3fff1ca0
3fffff90:  3fffdad0 00000000 3fff0ea8 40207189
3fffffa0:  feefeffe feefeffe 3fff1c8c 40215f4c
3fffffb0:  feefeffe feefeffe 3ffe8600 401013f9
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00060200
~ld
Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.50
Board ID: feather-esp8266
Adafruit.io User: K9LD
WiFi Network: Ma Pa Dalton 2.4GHz
MAC Address: C8:C9:A3:90:37:27
-------------------------------
Subscribing to MQTT topics...
Running Network FSM...
Attempting to connect to WiFi...
CONNECTING
FSM_NET_ESTABLISH_MQTT

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40222276 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  40106255 0049777a 3fff1ffc 00000000
3ffffeb0:  3fff0a40 3fff1ffc 0000012c 4020d98e
3ffffec0:  000000ff 3fff1ffc 3fff2eec 4020d8ae
3ffffed0:  4021731d 0000012c 00000014 40217312
3ffffee0:  00000005 0000012c 00000000 00000014
3ffffef0:  00000005 3fff18ac 3fff0ea8 40208581
3fffff00:  40214dc8 3ffeb180 3fff1c14 40215034
3fffff10:  00000001 00000016 3fff1c14 40215118
3fffff20:  000000a3 00000090 3fff19ac 00000000
3fffff30:  3fff18ac 3fff0ea8 3fff1c14 402088db
3fffff40:  00000037 00000027 ffffffff 00000001
3fffff50:  433a3843 33413a39 3a30393a 323a3733
3fffff60:  00000037 00000001 3fff1c14 40214e74
3fffff70:  00000000 00000000 3fff0ea8 3fff1ca0
3fffff80:  0001c200 0000001c 00000000 3fff1ca0
3fffff90:  3fffdad0 00000000 3fff0ea8 40207189
3fffffa0:  feefeffe feefeffe 3fff1c8c 40215f4c
3fffffb0:  feefeffe feefeffe 3ffe8600 401013f9
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00060200
~ld
Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.50
Board ID: feather-esp8266
Adafruit.io User: K9LD
WiFi Network: Ma Pa Dalton 2.4GHz
MAC Address: C8:C9:A3:90:37:27
-------------------------------
Subscribing to MQTT topics...
Running Network FSM...
Attempting to connect to WiFi...
CONNECTING

I don't see any revision increase in what I just downloaded. Looks like the same code causing the crash.
1945 est 22sep22

brentru, 2022-09-23:
@crustylarry This is with an Adafruit ESP8266 Feather?

crustylarry, 2022-09-23:
Feather HUZZAH ESP8266 WiFi. 4mb Flash @ 3.3v

brentru, 2022-10-07:
Also reported here: https://forums.adafruit.com/viewtopic.php?t=194727

brentru, 2022-10-13:
@crustylarry  I've been unable to replicate this issue with latest (beta 51) firmware and an ESP8266. I even tried your SSID to help replicate the problem.

Only other thing I can think of is - what's the length of your WiFi password, in characters?

brentru, 2022-10-13:
Ok, I'm able to replicate this issue by providing an incorrect Adafruit IO Key.

brentru, 2022-10-13:
Noting that the status LED is also not blinking on this board, needs to be fixed.

closed 2022-10-13 by brentru