adafruit/Adafruit_Wippersnapper_Arduino #335: New version of beta 50 released hours ago causes new bug in v2

TheGodMotherG, 2022-09-23:
**Describe the bug**
Updated with new version of 50 beta 

***Arduino board***
Esp32 v2 

**To Reproduce**
Steps to reproduce the behavior:
Install through web flash as always 

**Expected behavior**
With old version of 50 it was stuck in a boot loop restart 

Now it's just a solid red light and this is the error code

20:50:46.024 -> Fets Jul 29 2019 12:21:46
20:50:46.024 -> 
20:50:46.024 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:50:46.024 -> configsip: 271414342, SPIWP:0xee
20:50:46.024 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:50:46.024 -> mode:DIO, clock div:1
20:50:46.024 -> load:0x3fff0030,len:1184
20:50:46.024 -> load:0x40078000,len:13160
20:50:46.024 -> load:0x40080400,len:3036
20:50:46.024 -> entry 0x400805e4
20:50:46.633 -> ./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
20:50:46.633 -> E (236) esp_littlefs: mount failed,  (-84)
20:50:46.633 -> E (236) esp_littlefs: Failed to initialize LittleFS

**Which components are connected to your device**

Can't even get device to register haven't had anything 
 work since upgrade 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

brentru, 2022-09-23:
@TheGodMotherG  Could you try visiting https://adafruit.github.io/WipperSnapper_Firmware_Uploader/ and clicking "Update Credentials ONLY"?

brentru, 2022-09-23:
The Web Flasher is having conflicts with [the new ESP32 board support package release](https://github.com/espressif/arduino-esp32/releases/tag/2.0.5). We are working on resolving this issue.

TheGodMotherG, 2022-09-23:
I went to the adafruit firmware upgrader page. Just installed 50 again same
red light. Then unplugged reloaded page. And went to credentials only. Same
issue

TheGodMotherG, 2022-09-23:
When I look at serial monitor after upgrading then doing credentials only I
get this message

12:34:18.277 -> ⸮ets Jul 29 2019 12:21:46
12:34:18.277 ->
12:34:18.277 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:34:18.277 -> configsip: 271414342, SPIWP:0xee
12:34:18.277 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,
wp_drv:0x00
12:34:18.277 -> mode:DIO, clock div:1
12:34:18.277 -> load:0x3fff0030,len:1184
12:34:18.277 -> load:0x40078000,len:13160
12:34:18.323 -> load:0x40080400,len:3036
12:34:18.323 -> entry 0x400805e4
12:34:18.885 -> ./components/esp_littlefs/src/littlefs/lfs.c:4191:error:
Invalid block count (25 != 400)
12:34:18.885 -> E (236) esp_littlefs: mount failed,  (-22)
12:34:18.885 -> E (236) esp_littlefs: Failed to initialize LittleFS

brentru, 2022-09-23:
@TheGodMotherG  Please try the web uploader again, we've deployed a fix.

brentru, 2022-09-23:
Addressing all ESP32 web uploader boards in https://github.com/adafruit/Wippersnapper_Boards/pull/78, deploying and testing..

brentru, 2022-09-23:
All affected ESP32 boards have been patched and tested with the Web Uploader on Beta 50.

This does not resolve https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/334, which is a separate BSP.

Please feel free to re-open if you are experiencing the same issue.

closed 2022-09-23 by brentru

TheGodMotherG, 2022-09-23:
Thank you so much it works!!!!! You are amazing !!!!!!

I saw that The option for Fahrenheit was added. Is that in this release?
And if so how do you set it?

brentru, 2022-09-24:
It was not added yet, soon!

TheGodMotherG, 2022-09-25:
Thank you for your email can’t wait.