adafruit/Adafruit_Wippersnapper_Arduino #336: Improve PlatformIO Build Instructions in README.md

brentru, 2022-09-27:
The [PlatformIO build instructions in the README.md](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino#platform-io-build-instructions) could use some improvement:

1. Test the PlatformIO build instructions on your local machine, comment below with feedback on how it can be improved. We'll discuss more below!
2. Update the build instructions in the `README.md` to reflect feedback from discussion
3. Create a pull request with the updated `README.md` file

tyeth, 2022-10-16:
Hiya, I've tried running and the feather32v2 compiles but wont link.
[image]
The feather-s2 runs into a tinyUSB error maybe due to missing tiny_usb rename script (ref https://github.com/platformio/platform-espressif32/issues/809 ). 
[image]
The feather8266 builds and wont link
[image]
Running on windows 11 in a powershell / platformio terminal

brentru, 2022-10-18:
@tyeth  Here's what my current `.ini` looks like.

```
; PlatformIO Project Configuration File for Adafruit.IO WipperSnapper
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
description = "Adafruit.IO WipperSnapper"
default_envs = featheresp32v2, featheresp32s2

; Common WipperSnapper Build Environment
; https://docs.platformio.org/en/latest/projectconf/section_env.html
[env]
framework = arduino
monitor_speed = 115200
lib_compat_mode = strict
lib_extra_dirs = ~/Documents/Arduino/libraries

; WipperSnapper local build environment for Adafruit Feather ESP32 v2
[env:featheresp32v2]
platform = https://github.com/platformio/platform-espressif32.git
;platform = https://github.com/Jason2866/platform-espressif32.git
monitor_filters = time, esp32_exception_decoder, colorize
board = adafruit_feather_esp32_v2
build_flags = -D ARDUINO_ADAFRUIT_FEATHER_ESP32_V2
upload_port = /dev/cu.wchusbserial54790143761
monitor_port = /dev/cu.wchusbserial54790143761
lib_deps = adafruit/Adafruit TinyUSB Library@^1.10.3
; build_src_filter = +<*> -<../.pio/libdeps/featheresp32v2/MAX31850 OneWire>
;board_build.partitions = default_8MB.csv
;extra_scripts = ./littlefsbuilder.py

; WipperSnapper local build environment for Adafruit Feather ESP32-S2
[env:featheresp32s2]
platform = espressif32
board = featheresp32-s2
upload_port = /dev/cu.usbmodem01
monitor_port = /dev/cu.usbmodem1330
build_flags = -D ARDUINO_ADAFRUIT_FEATHER_ESP32S2
monitor_filters = time, esp32_exception_decoder, colorize
extra_scripts = 
    pre:rename_usb_config.py
lib_deps = adafruit/Adafruit TinyUSB Library@^1.10.3

; WipperSnapper local build environment for Adafruit Feather ESP8266
[env:featheresp8266]
platform = https://github.com/platformio/platform-espressif8266.git
board = huzzah
build_flags = -D ARDUINO_ESP8266_ADAFRUIT_HUZZAH
lib_ignore = Adafruit TinyUSB Library
build_src_filter = +<*> -<../.pio/libdeps/featheresp8266/Adafruit TinyUSB Library>
upload_port = /dev/cu.usbserial-017C8C08
monitor_port = /dev/cu.usbserial-017C8C08
build_type = debug
monitor_filters = esp8266_exception_decoder
```

> Hiya, I've tried running and the feather32v2 compiles but wont link for me.
Could you show me a screenshot of what your VSCode side-bar looks like, with all the files listed?

tyeth, 2022-10-21:
@brentru similar situation on a fresh installed pc (well actually Windows Sandbox), with vscode + platformio + git for windows.
Also there is a dallas ds18b20 dependency missing (it's silently cloned as part of the CI, but not listed in library.properties)

My local (not fresh) looks like this (I also have another clone/copy in linux / WSL with same situation - not fresh)
[image]

brentru, 2022-10-21:
@tyeth  This is good information. I haven't tried building from scratch but would like to..

>  so I ended up having to install the libraries to platform io with the pio lib install which is now deprecated. Is there a way to get platformIO to install requirements from library.properties? I had to do a regex replace and make a list of commands as I am under the impression each library must be installed individually. I tried pio pkg install and I think it just did tiny-usb

My understanding is that [this line](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/main/platformio.ini#L16) tells PIO to source from the standard Arduino library folder via [the PIO Lib Dep. Finder](https://bit.ly/configure-pio-ldf).

Output from my Build command, for reference
```
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 219 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit TinyUSB Library @ 1.14.3
|-- Adafruit WipperSnapper @ 1.0.0-beta.52
|   |-- Adafruit DotStar @ 1.2.1
...
```

> Also there is a dallas ds18b20 dependency missing (it's silently cloned as part of the CI, but not listed in library.properties)

Yep, expected. We may want to pull that into the platformio.ini for **local** builds.

What does doing a Build on the ESP32v2 platform look like for you?

tyeth, 2022-10-24:
Okay so the readme probably needs to say follow the arduino guide and install the required arduino libraries. Alternatively use the arduino-cli which can install all the libraries in one go and no arduino IDE needed. Lastly if you _only_ want platformio (no arduino CLI/IDE) then you can install the libraries mentioned in library.properties directly with platformIO.

The ds18b20 works fine for me (got one). It does need the libraries adding to the library.propties as currently the arduino guide/build is broken without.

brentru, 2022-10-24:
> Lastly if you only want platformio (no arduino CLI/IDE) then you can install the libraries mentioned in library.properties directly with platformIO. ( tail .\library.properties -n1 ).split('=')[1].split(',') | ForEach-Object { pio.exe lib install $_.trim() }

Where did you insert this snippet? Is this Powershell? You can execute python scripts from the platformio.ini file, maybe we'll want something like this in there?

tyeth, 2022-10-24:
Yeah powershell, ran in platformIO terminal, tried to do code highlight in markdown but it just colour-codes not labels.
<details>
<summary>My espV2 as requested builds (this is another laptop) but wont link</summary>
```
Executing task: C:\Users\tyeth\.platformio\penv\Scripts\platformio.exe run --environment featheresp32v2 

Processing featheresp32v2 (platform: https://github.com/platformio/platform-espressif32.git; board: adafruit_feather_esp32_v2; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_feather_esp32_v2.html
PLATFORM: Espressif 32 (5.2.0+sha.03254d4) > Adafruit Feather ESP32 V2
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 66 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit NeoPixel @ 1.10.6
|   |-- Adafruit TinyUSB Library @ 1.14.4
|-- Adafruit SPIFlash @ 4.0.0
|   |-- SPI @ 2.0.0
|   |-- SdFat - Adafruit Fork @ 2.2.1
|   |   |-- SPI @ 2.0.0
|-- ArduinoJson @ 6.19.4
|-- Adafruit DotStar @ 1.2.1
|   |-- Adafruit BusIO @ 1.13.2
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit SleepyDog Library @ 1.6.1
|-- Adafruit TinyUSB Library @ 1.14.4
|-- Adafruit AHTX0 @ 2.0.3
|   |-- Adafruit BusIO @ 1.13.2
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.6
|-- Adafruit BME280 Library @ 2.2.2
|   |-- Adafruit BusIO @ 1.13.2
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.6
|-- Adafruit DPS310 @ 1.1.1
|   |-- Adafruit BusIO @ 1.13.2
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.6
|-- Adafruit SCD30 @ 1.0.9
|   |-- Adafruit BusIO @ 1.13.2
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.6
|-- Sensirion I2C SCD4x @ 0.3.1
|   |-- Wire @ 2.0.0
|   |-- Sensirion Core @ 0.6.0
|   |   |-- Wire @ 2.0.0
|-- arduino-sht @ 1.2.2
|   |-- Wire @ 2.0.0
[... 57 lines cut ...]
|-- Wire @ 2.0.0
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- SdFat - Adafruit Fork @ 2.2.1
|   |-- SPI @ 2.0.0
Building in release mode
Linking .pio\build\featheresp32v2\firmware.elf
c:/users/tyeth/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/featheresp32v2/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
c:/users/tyeth/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/featheresp32v2/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
c:/users/tyeth/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/featheresp32v2/libFrameworkArduino.a(main.cpp.o): in function `loopTask(void*)':
C:/Users/tyeth/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42: undefined reference to `setup()'
c:/users/tyeth/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/tyeth/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:48: undefined reference to `loop()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\featheresp32v2\firmware.elf] Error 1
============================================================================================================== [FAILED] Took 79.38 seconds ==============================================================================================================

Environment     Status    Duration
--------------  --------  ------------
featheresp32v2  FAILED    00:01:19.378
========================================================================================================= 1 failed, 0 succeeded in 00:01:19.378 =========================================================================================================

```
</details>
I ended up adding DallasTemperature in platformIO as same author.

tyeth, 2022-10-31:
So I've got a branch with the dallas dependency, and the python script to setup platformio dependencies.
I still get the usb errors for the S2/S3 chipset builds, was trying to build the featherS2 (dont have one) and add the QTPY-S2, and feather-s3 (which I have) related to issue #337 
As far as this issue goes, I've got a python script which will install the dependencies mentioned in library.properties and I've made a suggested change to the readme. Let me know what you think - pull request #357

closed 2023-07-19 by brentru