adafruit/Adafruit_Wippersnapper_Arduino #338: Improve error handling during initial connection w/ WebUSB

brentru, 2022-09-28:
Originally an idea from @tannewt. 

We may want to force users to re-connect their boards to WipperSnapper during the initial registration process to allow IO to debug errors "for them". 

Error messages during this process would be piped over serial as JSON-formatted packets. IO would be able to process/understand the state of the board's JSON error/success packets and relay human-readable "next steps" to a user.

This could also be handled by the last step of the "New Device" workflow where there would be a "Still not connected?" option which would open this "debug mode".

It may be presented to the user as a "step" in the "New Device" workflow to make the process sit on one end of the site.

cc @lorennorman for technical feedback/comments

lorennorman, 2022-09-28:
I think I understand it, basically if the final reboot doesn't work to get the device online (the "happy path" fails), then the guidance is to plug it in directly and enter a debugging mode.

I like this! I think it:
- gets built AFTER we port the firmware tool into IO proper (it's still an external site right now)
- is a new, extra step after the final, "blinking lights" step of the happy path
  - there could be a small link for "Not connecting?" that's always visible, and...
  - a pop-up after 1 minute that says "Not working?" and explicitly invites them to the debug step
- is also a stand-alone page that can be directly accessed, without going through the board selector flow at all (seems generally useful, no need to lock it away at the end of a process)

brentru, 2023-09-06:
@lorennorman  We should revisit this next week

brentru, 2023-09-12:
More thoughts:
* Do we need a JSON stream or can we write out to a JSON file?
* Can we detect if the serial connect is open? If we can, the WDT can be disabled so it doesn’t reboot while the user is understanding and fixing their error?