adafruit/Adafruit_Wippersnapper_Arduino #339: SHT3X - Component doesnt exist in this firmware - v51

tyeth, 2022-09-28:
Fails to initialise component after selecting i2c address and sampling frequency.
Only Temp(c) and Humidity show up, when testing the SCD41 with the new firmware (Temp-Fahrenheit has been added) it should have an extra metric now for fahrenheit. I wonder if the website is just out of date/sync, but I cleared browser cache to be sure. The SHT40 shows fahrenheit, but the sht3x doesnt :'(
https://github.com/adafruit/Wippersnapper_Components/issues/74
```
Your I2C Component at address 0x44 could not initialize:

Error Message: WipperSnapper firmware is outdated and does not include the sensor type. Please update your WipperSnapper firmware.
```

brentru, 2022-09-28:
The SHT3x was added prior to us updating the sensor type..

[In the component definition for the SHT3x](https://github.com/adafruit/Wippersnapper_Components/blob/main/components/i2c/sht3x/definition.json#L4), add `ambient-temp-fahrenheit` to the component's `subcomponents` array.

See the SHT4x definition for a working example: https://github.com/adafruit/Wippersnapper_Components/blob/main/components/i2c/sht40/definition.json#L4

You do not need to do any extra modifications to the driver itself (within WipperSnapper_Arduino), the firmware should take care of it.

@lorennorman  We should do a sweep of the current components that support `ambient-temp` but not contain the `-fahrenheit` tag.

tyeth, 2022-09-29:
Thanks for the info, resolved in pull request on components repo and stuck in a readme fix for good measure.

closed 2022-09-29 by tyeth

tyeth, 2022-09-29:
Pre-emptive closure, although the fahrenheit metric was missing that gives a failed initialisation error instead of component missing from firmware:

Changes:
 The component name is not fully lowercase (sht3X) and the method name for ambient temperature was changed:
https://github.com/tyeth/Adafruit_Wippersnapper_Arduino/compare/add-sht3X...adafruit:Adafruit_Wippersnapper_Arduino:main#diff-0af6c374887eea60428bc26e217b466b5f576afc0794e220aa24f35d6eb5848dL74

I've stuck in a pull request (I tested the fix with and without each commit, both are required) that fixes the crux of the matter.