adafruit/Adafruit_Wippersnapper_Arduino #341: Bug: Two of same component breaks things

tyeth, 2022-09-29:
I'm leaving the title a bit loose to illustrate two issues I've found tonight.

 The first is that pin in use messages don't work.
 e.g. if I use a component (SHT40) on i2c address 0x44, and then try to add an SHT3X family component then it tells me address 0x44 is in use and only the remaining SHT3X addresses are available.
  If I instead have a first SHT3x sensor at address 0x44 and then try to add a second SHT3x sensor, the dialog does not detect / signal that 0x44 is already in use when adding the second instance. I can however use my brain and pick 0x45 manually. It is also annoying to have to rename each metric for multiple instances of the same sensor, so maybe an address suffix or auto-naming can be thought about.

  The second more major issue is I can't correctly edit the first instance after adding the second. I include a 10mb gif to illustrate, but basically add the 0x44 and then 0x45 (with a 5 suffix to metric names) then go and edit Temperature metric name for each sensor (you cant).

[image]

brentru, 2022-09-29:
This may be a firmware issue, transferring issue over...

brentru, 2022-12-02:
@lorennorman This is a reproducible bug from my end and requires a fix on the web:
> I instead have a first SHT3x sensor at address 0x44 and then try to add a second SHT3x sensor, the dialog does not detect / signal that 0x44 is already in use when adding the second instance.

brentru, 2022-12-02:
@tyeth 
> The second more major issue is I can't correctly edit the first instance after adding the second. I include a 10mb gif to illustrate, but basically add the 0x44 and then 0x45 (with a 5 suffix to metric names) then go and edit Temperature metric name for each sensor (you cant).

I just ordered another SHT31-D from Adafruit. Moving this bug's status to "Needs Investigation" as I wait for parts.

brentru, 2022-12-05:
@lorennorman  Both issues on this repo seem to be on the web-side of things rather than the WipperSnapper-side. They are both reproducible from my end as well.

lorennorman, 2022-12-05:
This is on the way to production now, will be live in 5 minutes.

Changes:
- I2C addresses should now properly reflect if they're already in use and unselectable
- all component names now default to non-duplicate names

I was unable to reproduce the issue where it never pulled up the appropriate form for the first of 2 I2C components of the same type. It is possible I fixed it with the first 2 fixes, but please test and re-open if not.

Thanks!

closed 2022-12-05 by lorennorman