adafruit/Adafruit_Wippersnapper_Arduino #343: Data stops pulling / plotting to feed after about 2 hours

videopixil, 2022-10-05:
**Describe the bug**
SCD30 data stops pulling after about two hours in WipperSnapper v1.0.0-beta.51

***Arduino board***
QTPY esp32-S2 

**To Reproduce**
Steps to reproduce the behavior:
1. add any sensor 
2. pull data for about two hours
3. notice after two hours device is online but not pulling / updating plot feed
4. no error shown

**Expected behavior**
should keep pulling data

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?
scd30 and stemma soil sensor

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Mac OS,  iOS
 - Browser safari
 - Version 16

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone14]
 - OS: [e.g. iOS16.0.2]
 - Browser safari
 - Version 16

**Additional context**
can pull data again if component is deleted and re added or if device is removed and re added

brentru, 2022-10-12:
> it just stops until i delete component and re add. when i switch to a new battery it shows it going back online but data stops then to

videopixil, 2022-10-20:
@brentru noting beta 47 and below were able to reconnect perfectly after powering down

brentru, 2022-10-20:
@videopixil Is it only on beta 50 and up?

videopixil, 2022-10-20:
@brentru it started in beta 49 and above - noticed in the canary project

brentru, 2023-04-13:
@videopixil  Does this still occur?