adafruit/Adafruit_Wippersnapper_Arduino #345: I2C not working on ESP32-S2 TFT

isomorphic85, 2022-10-07:
**Describe the bug**
I2C devices are not discoverable on ESP32-S2 TFT boards.

I have tried numerous different I2C sensors including several BME280 I2C sensors .

Manually trying to add a bme280 doesn't work either as it can't see the I2C device

***Arduino board***
ESP32-S2 TFT

**To Reproduce**
Connect I2c Sensor
1. Go to devices / I2C Scan
2. Click on Scan Again 
3. No devices will show 
4. Scan show's complete

**Expected behavior**
expecting to see the i2c device that is connected's address

**Which components are connected to your device**
none, unable to add any due to i2c not working

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: win10
 - Browser chrome
 - Version 105.0.5195.127 (Official Build) (64-bit)

**Additional context**
Tried with different ESP32-S2 TFT boards and all had same problem.

isomorphic85, 2022-10-10:
any update on this?

brentru, 2022-10-11:
This issue will be looked at this week as part of the batch of issues tagged for beta 52

https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/milestone/3

brentru, 2022-10-13:
Testbed I2C scan works on my board
> Default port I2C scan: 0x0, 0xB, 0x38, 0x77, 

WipperSnapper I2C scan is same as issue

brentru, 2022-10-13:
Hi @isomorphic85  - I've resolved this issue. You do not need to update the firmware! There was a mismatch between the [I2C pins on the definition](https://github.com/adafruit/Wippersnapper_Boards/commit/b93ba14b747acb9588ce9c94de97a23daad78569) we use and the actual I2C pinout. This has since been fixed.

Please try **deleting** the ESP32-S2 TFT Feather from your Adafruit IO devices page. Then, refresh your tab and reconnect your board. Please let me know if it's working for you. I tested it as well with a similar setup:

I2C scan with BME280 and AHT20 attached to ESP32-S2 TFT's I2C Port:
[image]

BME280 attached to ESP32-S2 TFT
[image]

isomorphic85, 2022-10-13:
That worked!  Awesome.

Thank you

brentru, 2022-10-14:
> side note- any support for images or data on the screen in the works through wipper?

Like on a TFT? Yes, it's on the roadmap.

closed 2022-10-14 by brentru