adafruit/Adafruit_Wippersnapper_Arduino #346: error adding bme680, tsl2591 to a qtpy 32s2

videopixil, 2022-10-07:
**Describe the bug**
error adding bme680, tsl2591 to a qtpy 32s2 - running beta 51 says firmware out of date

***Arduino board***
qtpy 32s2 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'add component'
2. Click on 'bme680, tsl2591 or bme280'
3. See i2c error firmware out of date

**Expected behavior**
component should add

**Which components are connected to your device**
none

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS 12.6
 - Browser safari
 - Version 16

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: 16

brentru, 2022-10-14:
@videopixil  Driver for BME680 was never added, ref: https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/262. I'll add it.

I'll take a look at the TSL issue as well. I'll tag you in when I open a PR with a UF2 file to test with.

brentru, 2022-10-20:
@videopixil  This issue is resolved via https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/355 and will be released with Beta 53 later today. Please test this and let me know if it works for you.

[image]

videopixil, 2022-10-20:
@brentru awesome will test as soon as its live!

closed 2022-10-20 by brentru