adafruit/Adafruit_Wippersnapper_Arduino #354: Setup on Funhouse to read AHT20 sensor to Fahrenheit reverted back to Celsius

kspray, 2022-10-19:
**Describe the bug**
A clear and concise description of what the bug is.
I started with beta .51 and setup the Funhouse with an AHT sensor to read and display Fahrenheit temperature every 5 minutes. I noticed the first couple of readings were in Celsius, then readings 5 and 6 were in Fahrenheit, the the rest reverted back to Celsius. I updated to beta .52 and I am still getting the same Celsius readings. I've also deleted the setup and reconfigured, but it appears to still be reading in Celsius. I am unable to get it to recreate the reading Celsius then Fahrenheit, then Celsius, but it does still continue to read in Celsius when I have set it up for Fahrenheit.
***Arduino board***
INSERT ARDUINO BOARD NAME/TYPE HERE
Adafruit Funhouse ESP32-S2
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '... Devices and click on the  Funhouse device, click on new component and scroll down and select the AHT20 component. Uncheck the box next to "Enable AHT20: Temperature Sensor (°C)?". I changed the send every dropdown for both the temperature and humidity readings to every 5 minutes and then finalized by clicking create component button

**Expected behavior**
A clear and concise description of what you expected to happen.
I expected to get a graph and table with Fahrenheit readings. 
**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?
Funhouse and AHT20 sensor
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome Version 106.0.5249.119 (Official Build) (64-bit)

**Additional context**
Add any other context about the problem here.

brentru, 2022-12-01:
@kspray 

>  setup the Funhouse with an AHT sensor to read and display Fahrenheit temperature every 5 minutes. I noticed the first couple of readings were in Celsius, then readings 5 and 6 were in Fahrenheit, the the rest reverted back to Celsius.

I'm unable to replicate this so far on Beta 55. I have my funhouse logging and will keep it online overnight to check. Is this issue still occurring for you?

kspray, 2022-12-15:
It's working right now. you can close this. I'm embarrassed. I must have not had the sensor plugged in all the way or somehting.

brentru, 2022-12-15:
Great!

closed 2022-12-15 by brentru