adafruit/Adafruit_Wippersnapper_Arduino #356: Wippersnapper won't initialize VL53L0X sensor

ETJumper, 2022-10-23:
When trying to add a VL53L0X I2C sensor, using the guide on Adafruit, I keep getting an error that I2C device failed initialization and the following message: 
"Your I2C Component at address 0x29 could not initialize: 
Error Message: Wippersnapper firmware is outdated and does not include the sensor type. Please update your Wippersnapper firmware."

Board is an Adafruit QT py ESP32 Pico.  Using Chrome browser Version 106.0.5249.119 (Official Build) (64-bit) on a Window Surface Laptop 3 running Windows 10 Pro.

I've tried removing and reinstalling the device (QT Py ESP32 Pico) with WipperSnapper. I2C Scan shows correct address. VL53L0X Sensor tested good with a regular QT Py (SAMD21).  Even tried rebooting the computer.  The QT Py will take a program (Blink) using the Arduino IDE.

brentru, 2022-10-24:
I believe https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/291 was not completed by the original author and thus is not supported by WipperSnapper firmware despite being on the icon picker. We're working on a solution to hide these types of issues in the coming week.

You're welcome to either pick up this pull request (we have a guide on adding components here: https://learn.adafruit.com/how-to-add-a-new-component-to-adafruit-io-wippersnapper/adding-an-i2c-component-driver) or wait for us to add it to WipperSnapper.

tyeth, 2022-11-12:
I'm going to push this forwards as my VL53L0X (and 1X) sensors have arrived

brentru, 2022-11-28:
Addressing in https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/363

tyeth, 2022-12-08:
@ETJumper give the latest version a try if you're still interested. It does return sensor data. 
Currently you'll need to select the checkbox to "Show Dev?" which will show components still under development:
[image]

## Notice the orange banner:

[image]

I think this ticket can probably be closed and then bugs raised if necessary.

brentru, 2022-12-08:
@tyeth  Do we still want this as an "in development" component, as it is now in Production? I can make it live if so.

tyeth, 2022-12-08:
Make it so captain

closed 2022-12-08 by brentru

tyeth, 2022-12-08:
Sweet, it's there. No refreshing, but it was a new tab.