adafruit/Adafruit_Wippersnapper_Arduino #364: Better build instructions/environment for local compilation of WipperSnapper Arduino library

brentru, 2022-11-14:
@lorennorman  was having issues getting WS Arduino compiled locally. @tyeth was as well, but with PlatformIO.. I'd prefer not to have a complex docker setup for this, but a makefile/shell script would be OK.

Making a list of enhancements and issues along the way...

Pain Points I noticed:
* Arduino IDE
  * Installing BSPs
  * Installing all dependencies 
  * Requires knowledge of the Arduino IDE
  * Sln: Switching to [Arduino-CLI ](https://github.com/arduino/arduino-cli) in our documentation
  * Sln: Better instructions around PlatformIO usage
  * Sln: Local usage of [Adafruit's CI-Arduino](https://github.com/adafruit/ci-arduino) tool
* Local modification/installation of Arduino libraries
  *  Documentation neglects to mention where libraries are located, how to test parts of WipperSnapper individually

brentru, 2023-01-26:
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/357

tyeth, 2023-01-26:
> #357

Still need to add DallasTemperature library to library.properties, but i've updated the added instructions branch that I did.

brentru, 2024-08-09:
Closing this as its handled by the learn guide

closed 2024-08-09 by brentru