adafruit/Adafruit_Wippersnapper_Arduino #365: Ability to turn off updating light.

MoonGoddess, 2022-11-15:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
Wishing to use Adafruit io sensor in a bedroom, however the code automatically flashes the neopixel very bright green when it updates, disturbing sleep.

**Describe alternatives you've considered**
Black tape over the neopixel, and putting the whole device in a box. (however the second solution interferes with the accuracy of air tempature readings.

**Additional context**
Just some sort of setting, no update light or no update light between these hours would be great!

MoonGoddess, 2022-11-15:
That would be ok,

honestly though, I was really expecting the adafruit.io website to be a lot more functional than it turned out to be.

brentru, 2022-11-15:
Ok, I'll look at adding it. WipperSnapper is still in beta + we are collecting user experiences...

What were you expecting to do with it? What can you do that you wanted to? What can you not do?

MoonGoddess, 2022-11-15:
> What can you not do?

Well, send information (current value of the feeds, time) to the TFT on the board, I had wished that I might be able to set up a button to do this so that wasn't on all the time, the ability to monitor the remaining battery on the board would also be nice.

brentru, 2022-11-15:
> Well, send information (current value of the feeds, time) to the TFT on the board, 

This will be added in time, https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/208, WipperSnapper is still in beta. This is one of the next features we're working on. You can track its status via this issue -> https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/208

You are interested in having the TFT also display the local time?

> I had wished that I might be able to set up a button to do this so that wasn't on all the time, t

So that the TFT is not on all the time?

> the ability to monitor the remaining battery on the board would also be nice.
We **do** support the LC709203 (the battery charging IC on the S2 TFT) - go into I2C Components and click on the LC709203.

MoonGoddess, 2022-11-15:
>  This will be added in time, https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/208, WipperSnapper is still in beta. This is one of the next features we're working on. You can track its status via this issue -> https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/208

Following

> You are interested in having the TFT also display the local time?

Yes,  not vital but a nice feature.

So that the TFT is not on all the time? 

Yes that is the desired result, as stated above, Bedroom

> We do support the LC709203 (the battery charging IC on the S2 TFT) - go into I2C Components and click on the LC709203.

Ahhh good to know.   the way the icons show it made me thing the only things supported were little addon boards.

tyeth, 2022-12-05:
First time it affected me last night, I'd love to be able to reduce brightness to 5% or something adjustable now that you've worked on PWM, off would be okay but I get reassured about logging by the lights during daytime (had a few stop updating feeds).  Maybe I just switch off bedroom ones only instead. It did just occur to me maybe I could setup an LED output and set to Status LED / backlight Pin to allow control from webpage.

Also the lipo i2c chip is on the Feather S3 with psram and no tft, seems working fine!
 The only warning about supported non-adafruit sensor things is they must be connected to the expected i2c port, which is the stemmaQT port if the board has one. More likely than not you'll be using a breakout board with Qwiic connector or soldering cables on. 
Here's my latest batch of "adaptors", dupont header joint to qwiic cable, but save yourself time and buy some :)
[image]

On the TFT interface subject, what would you expect/desire to see? Personally I want to see latest values + errors to identify flaky sensors/boards/wiring, an error count (last XXhrs) might be fine but need IO page to help investigate errors / logs (like data not updating but device online). Maybe boot shows IP etc, but the received config is the useful bit, and then each reading.
 Like the Status light being off, I'd be inclined to have the TFT optionally off after bootup sequence until the user button (Boot0) is pressed, waking the screen and staying on for 30secs, leaving it on permanently is more like a school demo / clock but the electricity bean-counters wont like it (plus screen burn).

brentru, 2022-12-05:
> First time it affected me last night, I'd love to be able to reduce brightness to 5% or something adjustable now that you've worked on PWM, off would be okay but I get reassured about logging by the lights during daytime (had a few stop updating feeds)

We could store a custom brightness default within the `settings.json` file.

> On the TFT interface subject...

I opened a new issue for this so we can consolidate ideas! https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/382

> The only warning about supported non-adafruit sensor things is they must be connected to the expected i2c port, which is the stemmaQT port if the board has one. More likely than not you'll be using a breakout board with Qwiic connector or soldering cables on.

Could you file another issue for this. We'd like to support this down the line.

tyeth, 2022-12-05:
> We could store a custom brightness default within the `settings.json` file.

Good plan, does it have any "optional" elements yet?

> Could you file another issue for this. We'd like to support this down the line.

https://github.com/adafruit/Wippersnapper_Boards/issues/70 
Should I duplicate it here?

> I opened a new issue for this so we can consolidate ideas! #382

❤️

brentru, 2022-12-05:
> Good plan, does it have any "optional" elements yet?

Yeah, `io_url` sets the MQTT broker address. 

Duplicate it there.

closed 2022-12-06 by brentru