adafruit/Adafruit_Wippersnapper_Arduino #376: Allow multiple network SSIDs to be defined within secrets.json file

brentru, 2022-11-29:
Allow for multiple SSIDs to be defined within secrets.json, by priority
```
"network_type_wifi":[
          {
		"network_ssid":"my_house",
		"network_password":"password"
	},
        {
		"network_ssid":"the_garden",
		"network_password":"password"
	}
]
```

via @tyeth  on https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/328#issuecomment-1253956614
> I think long term I'd love the support of single object and/or array for networks, and to potentially add a key for priority, or potentially just use RSSI based priority upon start/disconnect. i.e. currently it's:

isomorphic85, 2023-01-23:
YES, this would be amazing

brentru, 2024-06-24:
This should be simple with the new WiFiMulti library inside BSP 3.x

https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html#wifimulti

brentru, 2024-06-24:
@tyeth  Want to tackle this one next week?

tyeth, 2024-07-29:
This has been added under the new additional key of `alternative_networks`.

Please see this playground note for details:
https://adafruit-playground.com/u/tyeth/pages/using-multiple-wifi-network-credentials-in-adafruit-io-wippersnapper-firmware

closed 2024-07-29 by tyeth