adafruit/Adafruit_Wippersnapper_Arduino #381: feather s3 - issue early beta56

tyeth, 2022-12-05:
Probably my fault, but the s3 feather with psram is not keeping the co2 data updated, seems like it's logging erroneously too, this is the bootup and I've never seen so many connected messages between pings before:
[image]
Running on the vlx53l0x branch / build assets.

tyeth, 2022-12-05:
I've stuck it on adafruit/main and that connected message is gone, but I'm getting no data. Presumably due to a decode fail, I wonder if I got some legacy metadata I'm unaware of on that device. In the past failures on this and other devices I usually delete the sensors only. I'll give it a day and delete the device tomorrow.
[image]

tyeth, 2022-12-05:
It i2c scans fine 
[image]
[image]

brentru, 2022-12-05:
> It had a vlx53l0x and SCD41 connected, 

Do you know whether the VLX53l0x or the SCD41 is causing the sync packet error to be thrown?

tyeth, 2022-12-05:
I also wonder if I'm overloading the world, i.e. filling and overflowing the 30second timeperiod between polling, as I have about 8metrics being polled and some fail I guess a bit like #215 taking longer than normal. How do you get that packet logging level of verbosity, just turn up arduino log level? (WS_Debug is defined by default)

brentru, 2022-12-05:
I don't have the vlx but just tested the SCD41 resync and it was fine. Will wait on your results but this isn't blocking beta 56.

>  How do you get that packet logging level of verbosity, just turn up arduino log level? (WS_Debug is defined by default)

Enable debug mode within Adafruit MQTT Library

tyeth, 2022-12-06:
It's more interesting that I thought, seems to be too big packet, then some decode error. Not examined in detail, just got logging terminal to disk with auto-reconnect working. Here's the problem moment on startup 
[image]
And here's the log showing from a restart, and there's another forced restart at line 501 to see the repeat.
[featherS3_capture_2022-12-06_1356 - Copy.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/10167105/featherS3_capture_2022-12-06_1356.-.Copy.txt)

brentru, 2022-12-06:
> It's more interesting that I thought, seems to be too big packet, then some decode error.

Looks like the packet is 58 bytes, the wippersnapper's mqtt rx buffer is ~ 256 bytes. 

Could you take a screenshot of the device page on Adafruit IO so I can replicate your setup?

tyeth, 2022-12-06:
> Looks like the packet is 58 bytes, the wippersnapper's mqtt rx buffer is ~ 256 bytes.

Yeah decode error is 56 bytes, but the buffer length error is after init at the top, still only 201bytes, I wondered if multi-part messages are supported, or if I could have some old device metadata saved. I've not deleted this device yet.

brentru, 2022-12-06:
@tyeth  in Arduino IDE, go to Tools->Debug Level->Verbose. Does a SSL buffer error happen there?

tyeth, 2022-12-07:
Don't see anything related, surprised there's not more debug listed, I put it on Verbose mode.
[capture.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/10179116/capture.txt)

tyeth, 2022-12-12:
So I'm seeing similar on the QTPYs2. I'm unsure if it's switching versions from new to old leaving bad message hanging around, or related to power-cycling. 
 I stick a copy on, lets say 57, delete all the sensors and then create them again, without fahrenheit, then verify data flows. I create a dashboard for temp/humdity, and set the dashboard (and therefore feeds) to public. I then stick the device on a usb powerbank and no data flows. I stick the device on the laptop and it tells me decode packet error. 
I also sometimes delete the device entirely before adding sensors again. I stick an older firmware on, but long story short at some point I cock up a device and then it wont survive a power cycle. 
 While on the powerbank it responds to i2c scan, so I have removed and readded sensors and seen data flow, until replugging usb cable.

None of my testing has been methodical so will revisit when have patience.

brentru, 2022-12-12:
> if it's switching versions from new to old leaving bad message hanging around

Could be a possible reason, but, we didn't change any protocol buffer messages recently. cc @lorennorman ^

What USB powerbank do you have? Were you using it with beta 56?

tyeth, 2022-12-12:
powerbank or different mains->usb outlets, just not usb socket with serial listening

lorennorman, 2022-12-12:
I can't think of a way that our protobufs could be changing meaningfully, for these scenarios. We always use backwards compatibility if we make breaking changes and we haven't actually made a breaking change like that in awhile.

brentru, 2022-12-13:
Will take a full look at this after PicoW support is merged in.

brentru, 2022-12-13:
Alright, duplicated the packet issue

```
11:07:23.882 -> Signal Msg Tag: Pin Configuration
11:07:23.882 -> Initial Pin Configuration Complete!
11:07:23.882 -> * NEW MESSAGE [Topic: Signal-I2C]: 
11:07:23.882 -> 70 bytes.
11:07:23.882 -> ERROR: Unable to decode I2C message
11:07:23.882 -> Publishing to pin config complete...
11:07:24.408 -> Dropped a packet
11:07:24.938 -> Hardware configured successfully!
```

with the following setup:
[image]

Removing the LC7x sensor from this config does not yield the "Unable to decode I2C message" error, will look more into it. Possible packet size problem.

brentru, 2022-12-13:
Looks like the max. packet size got changed back to 150 bytes from 512 in commit https://github.com/brentru/Adafruit_MQTT_Library/commit/28435eaf58d85b3c30e55b965a73447d129c7774#diff-240d52c08217f01fd327187461355707b15e7e24e0e6e3da37f0a67fa12006ceL110

tyeth, 2022-12-13:
fascinating, well spotted! Oh and great work on the picow, just about to stick a copy on now 🥇

closed 2022-12-13 by brentru