adafruit/Adafruit_Wippersnapper_Arduino #383: Add Air Quality Index (AQI) 'sensor fusion' to the PM2.5 sensor

tylerdcooper, 2022-12-06:
Using the same setup as [this guide](https://learn.adafruit.com/diy-air-quality-monitor/code-walkthrough), it would be nice if the [PM2.5 sensor](https://www.adafruit.com/product/4632) gave you the option to show AQI.

It is more human readable, and much easier to understand than a simple PM value. Would be nice to have both the AQI value and AQI category (Unhealthy, Healthy, etc).

tyeth, 2022-12-08:
I love the AQI, but as each metric is often required to be over a 24hr average, and related method varies per country, I'm not sure what makes sense. I'm from the UK, but previously was using the American model for a basic on-device webpage.

 This bit highlights the issue for the USA AQI (the UK is just above and similar): https://en.wikipedia.org/wiki/Air_quality_index#Computing_the_AQI:~:text=For%20example%2C%20calculation%20of%20the%20ozone%20AQI%20requires%20computation%20of%20an%208%2Dhour%20average%20and%20computation%20of%20the%20PM2.5%20or%20PM10%20AQI%20requires%20a%2024%2Dhour%20average

brentru, 2022-12-08:
@tyeth So, calculating an AQI without averaging will result in higher-than-expected AQI values compared to the 24-hour rolling average. If you and other users are OK with this, I don't mind adding it as a type. 

Otherwise - we'd need to have this calculated by IO (as IO is storing the values) over the period.

tyeth, 2022-12-08:
Yeah to be honest as a "clean air guide" it's very useful, stick it on the list!

tyeth, 2023-11-24:
Update noting that there's an open PR waiting for changes on the PM25AQI repo, if no changes forthcoming I'll modify and merge. See https://github.com/adafruit/Adafruit_PM25AQI/pull/12