adafruit/Adafruit_Wippersnapper_Arduino #389: VL53L0X publishes but with error every time

tyeth, 2022-12-13:
**Describe the bug**
Just running latest version on the PicoW, using latest version of earls core for pico, loaded in uf2 mode from arduino ide v2.0.3 on mac.
 Added vl53l0x and scd41, was seeing random intermittent errors on the scd41 metrics, but interspersed with regular (every time) errors before the distance metric, which also lacks a newline before PUBLISHING. 
All metrics are published successfully.

***Arduino board***
PicoW

**To Reproduce**
Steps to reproduce the behavior:
1. Add vl53l0x and watch serial monitor
2. Observe similar to this in console:
```
ERROR: Failed to get proximity sensor reading!
Sensor 0x29
	Proximity: 238.00PUBLISHING -> I2C Device Sensor Event Message...
PUBLISHED!
PING!
```

**Expected behavior**
Console outputs no errors, something like 
```Sensor 0x29
	Proximity: 238.00
PUBLISHING -> I2C Device Sensor Event Message...
PUBLISHED!
```

**Which components are connected to your device**
VL53L0X + SCD41

I've turned on debugging and it appears to be working fine, no error is generated from the driver. It looks like an artifact of the data[0] instead of distance metric, which is expected per the original pull request but maybe wippersnapper is also checking the distance/proximity event type first hence the message.

brentru, 2022-12-13:
I don't have a vl53l0x for testing, is it specific to this board or firmware? Or the sensor?

tyeth, 2022-12-13:
Sensor type i think. Any distance sensor and I thought there were others, but I mistakenly gained that from the age old comments in someone elses PR, and now read it was just not available yet: 
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/289#discussion_r938273490

brentru, 2023-01-26:
@tyeth  Was this resolved by you? I can order a vl53l0x for next week.

tyeth, 2023-01-26:
No sorry, it involved resolving proximity type outputs which look like they should be setting a sensor type property called proximity but are currently just data[0], linked in that issue comment, and I got well distracted with tasmota.

Order one, or maybe better yet the 1X which is nearly identical and 4m range instead of 2m.

   Also consider nabbing yourself the VL53L5CX, it's the big brother, an 8x8 grid with 3D gesture sensing, has onboard processing but needs a small gesture firmware pushing to it though as part of driver initialisation so not ideal for small firmwares. It looks like there was one in the works that may have slipped,
   https://blog.adafruit.com/2021/09/14/vl53lcx-stemma-qt-breakout-design/
 I got the mini version from sparkfun, super cute and practical (**Warning** _No level shifting - its 3.3v_ )
   https://www.sparkfun.com/products/19013

brentru, 2023-01-27:
Getting distracted is never a bad thing :) 

Ordered!

brentru, 2023-09-06:
@tyeth Does this sensor still exhibit this behavior?

tyeth, 2023-09-08:
Just reconnected it to an S3 devkit-c and it's returning data without errors in general, but the first event after adding the component showed the error message followed by successful reads.
[image]

Doing a reboot showed no errors, another reboot showed two initial errors then good reads.
It's not every read at all, and could well be valid responses (the sensor can return failure to read). I've got it on my list to have a proper look another time. Closing for now

closed 2023-09-08 by tyeth