adafruit/Adafruit_Wippersnapper_Arduino #390: Support MCUs with built-in temperature sensors

brentru, 2022-12-14:
Microcontrollers like the RP2040 and [ESP32-S2/S3, ESP32-C2/C3 ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/temp_sensor.html) contain built-in temperature sensors. These are not ambient temperature sensors, they measure the temperature of the chip itself. 

This temperature returned by these sensors is not very accurate (low resolution and generally higher than ambient temperature) but would serve as a quick way to get temperature data into IO without the use of external parts.

cc @lorennorman  - This will likely need to be part of WipperSnapper_Components and WipperSnapper_Boards, too. Since it's _technically_ an ADC measurement, we could denote it as an analog pin component. Only certain microcontrollers have a sensor like this, we may want to figure out how to add this (possibly as a new value) to the `board.json` too.

References:
* https://github.com/espressif/esp-idf/tree/master/examples/peripherals/temp_sensor
* https://arduino-pico.readthedocs.io/en/latest/analog.html#float-analogreadtemp

lorennorman, 2022-12-15:
Oh this is fun! I can't envision the whole thing myself, it would help me to see a draft of what new JSON things we'd add to boards/components repos, and how it might show up in the component picker. From there I should be able to collaborate on that design and guess at a work estimate for this.

brentru, 2022-12-30:
Proposal for support within WipperSnapper Boards: https://github.com/adafruit/Wippersnapper_Boards/issues/94

brentru, 2022-12-30:
Proposal for support within WipperSnapper Components: https://github.com/adafruit/Wippersnapper_Components/issues/111