adafruit/Adafruit_Wippersnapper_Arduino #391: WipperSnapper data no show

theleestarr, 2022-12-18:
**Describe the bug**
using feather esp32-s2 and a qtpy, neither consistently report data to IO. for the most part, no data makes it to IO. the boards just pings and blinks.

***Arduino board***
 feather esp32-s2 and a qtpy-s2

**To Reproduce**
Follow wizard steps to setup new device

**Expected behavior**
There should be data from the sensors. 

**Which components are connected to your device**
i have tried bme680, scd41, bme280, pm2.5

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 108.0.5359.99

**Smartphone (please complete the following information):**
none used

**Additional context**
serial monitor of board start.
20:34:35.292 -> Attempting to connect to Adafruit IO...
20:34:37.111 -> Registering hardware with WipperSnapper...
20:34:37.111 -> Registering hardware with IO...
20:34:37.111 -> Encoding registration request...Encoding registration msg...Published!
20:34:37.205 -> Polling for registration message response...2
20:34:37.613 -> GOT Registration Response Message:
20:34:37.613 -> Hardware Response Msg:
20:34:37.613 -> GPIO Pins: 11
20:34:37.613 -> Analog Pins: 6
20:34:37.613 -> Reference voltage: 2.60v
20:34:37.707 -> Completed registration process, configuration next!
20:34:37.707 -> Polling for message containing hardware configuration...
20:34:37.707 -> Polling for message containing hardware configuration...
20:34:37.754 -> Polling for message containing hardware configuration...
20:34:37.754 -> cbSignalTopic: New Msg on Signal Topic
20:34:37.754 -> 2 bytes.
20:34:37.754 -> decodeSignalMsg
20:34:37.754 -> cbSignalMsg
20:34:37.754 -> Sub-messages found: 1
20:34:37.754 -> Signal Msg Tag: Pin Configuration
20:34:37.754 -> Initial Pin Configuration Complete!
20:34:37.754 -> Publishing to pin config complete...
20:34:37.754 -> * NEW MESSAGE [Topic: Signal-I2C]:
20:34:37.754 -> 71 bytes.
20:34:37.754 -> ERROR: Unable to decode I2C message
20:34:37.754 -> Dropped a packet
20:34:37.754 -> Dropped a packet
20:34:37.848 -> Dropped a packet
20:34:38.318 -> Hardware configured successfully!
20:34:40.193 -> Registration and configuration complete!
20:34:40.193 -> Running application...
20:34:40.193 -> PING!
20:34:44.782 -> PING!
20:34:49.391 -> PING!
20:34:54.007 -> PING!

brentru, 2022-12-20:
@theleestarr Which version of WipperSnapper are you using?

theleestarr, 2022-12-21:
confirmed this is the error while running beta.57

tyeth, 2023-01-08:
Try less metrics / sensors, i.e. just one. When I have only 3 or 4 metrics it's fine, when I goto 11 it blows up on next reboot with decode error and just pings away happily doing nothing. The particle sensors add quite a few metrics in one go (my SEN55 is 3xPPM, 3xtemp/humidity and two VOC/NOx indexes), and temp/humidity sensors are worth 3 (celcius/fahrenheit/humdity).

closed 2023-06-30 by brentru