adafruit/Adafruit_Wippersnapper_Arduino #396: WIPPERSNAPPER Lib for Arduino compilation issue

paolopenna, 2023-01-02:
**Describe the bug**
Adafruit **Wippersnapper_NoFS** example stops the compilation with errors.

***Arduino board***
Board is NODEMCU ESP8266. Adafruit HUZZA ESP8266 does the same.

**To Reproduce**
Select board: NODEMCU 1.0 (ESP-12E).
Load Example: Adafruit Wippersnapper_NoFS.
Click on VERIFY.
Compile stops with: 
```
C:\ADAFRUIT\Wippersnapper_NoFS\Wippersnapper_NoFS\Wippersnapper_NoFS.ino:38:68: error: no matching function for call to 'Wippersnapper_ESP8266::Wippersnapper_ESP8266(const char [9], const char [33], const char [8], const char [16])'
   38 | Wippersnapper_WiFi wipper(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);
      |                                                                    ^
In file included from c:\Users\paolo\Documents\Arduino\libraries\Adafruit_WipperSnapper\src/Wippersnapper_Networking.h:25,
                 from C:\ADAFRUIT\Wippersnapper_NoFS\Wippersnapper_NoFS\Wippersnapper_NoFS.ino:3:
```

**Expected behavior**
End up compilation with no error.

**Which components are connected to your device**
None.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - Tested OS: WIN7 x64, WIN 10 x64, WIN11 x64.
 - Browser N/A
 - Arduino IDE 2.0.3

**Additional information**
The full error report:

```
C:\ADAFRUIT\Wippersnapper_NoFS\Wippersnapper_NoFS\Wippersnapper_NoFS.ino:38:68: error: no matching function for call to 'Wippersnapper_ESP8266::Wippersnapper_ESP8266(const char [9], const char [33], const char [8], const char [16])'
   38 | Wippersnapper_WiFi wipper(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);
      |                                                                    ^
In file included from c:\Users\paolo\Documents\Arduino\libraries\Adafruit_WipperSnapper\src/Wippersnapper_Networking.h:25,
                 from C:\ADAFRUIT\Wippersnapper_NoFS\Wippersnapper_NoFS\Wippersnapper_NoFS.ino:3:
c:\Users\paolo\Documents\Arduino\libraries\Adafruit_WipperSnapper\src/network_interfaces/Wippersnapper_ESP8266.h:65:3: note: candidate: 'Wippersnapper_ESP8266::Wippersnapper_ESP8266()'
   65 |   Wippersnapper_ESP8266() : Wippersnapper() {
      |   ^~~~~~~~~~~~~~~~~~~~~
c:\Users\paolo\Documents\Arduino\libraries\Adafruit_WipperSnapper\src/network_interfaces/Wippersnapper_ESP8266.h:65:3: note:   candidate expects 0 arguments, 4 provided
c:\Users\paolo\Documents\Arduino\libraries\Adafruit_WipperSnapper\src/network_interfaces/Wippersnapper_ESP8266.h:49:7: note: candidate: 'Wippersnapper_ESP8266::Wippersnapper_ESP8266(const Wippersnapper_ESP8266&)'
   49 | class Wippersnapper_ESP8266 : public Wippersnapper {
      |       ^~~~~~~~~~~~~~~~~~~~~
c:\Users\paolo\Documents\Arduino\libraries\Adafruit_WipperSnapper\src/network_interfaces/Wippersnapper_ESP8266.h:49:7: note:   candidate expects 1 argument, 4 provided
Multiple libraries were found for "DallasTemperature.h"
  Used: C:\Users\paolo\Documents\Arduino\libraries\DallasTemperature
  Not used: C:\Users\paolo\Documents\Arduino\libraries\MAX31850_DallasTemp
Multiple libraries were found for "Servo.h"
  Used: C:\Users\paolo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\Servo
  Not used: C:\Users\paolo\AppData\Local\Arduino15\libraries\Servo
exit status 1

Compilation error: no matching function for call to 'Wippersnapper_ESP8266::Wippersnapper_ESP8266(const char [9], const char [33], const char [8], const char [16])'

```

brentru, 2023-01-03:
@paolopenna  Please use the https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/main/examples/Wippersnapper_demo/Wippersnapper_demo.ino sketch with the nodemcu 8266 board, instead of the noFS sketch. 

Then, once compiled and uploaded, visit https://adafruit.github.io/WipperSnapper_Firmware_Uploader/?board=feather-huzzah-esp8266 and select update credentials from the dropdown to upload the secrets.json to your ESP-8266-based board

Let me know if that compiles.

paolopenna, 2023-01-06:
Hello Brent and thank You for the answer.
Compile is OK, but from my understanding the next step is connecting through the web consolle and downloading the firmware. with this step, with the original HUZZA8266 also, i got stuck on a "TRY HARD RESET" . That's why I choosed the programmer way.
Any advice?
It should be said that with the HuzzaH32 the web consolle worked only on a WIN10 laptop, after some try, the issue was the same "try hard reset" stuck.
Thank You!
Paolo.

[image]