adafruit/Adafruit_Wippersnapper_Arduino #403: PICO_W Ain aka POTI no data

kllsamui, 2023-01-22:
**Describe the bug**
device config questionable and NO data

***Arduino board***
PICO_W

**To Reproduce**
+ device POTI
+ select GP0
+ see A0, NO data

**Expected behavior**
want select  GP26 27 28 for A0 A1 A2  OR GP29 for CPU TEMP

**Which components are connected to your device**
POTI 10k  3v3 to 0
slider to GP0 or GP26 ...
voltage confiirmed DMM

**Screenshots**

**Desktop (please complete the following information):**
 - OS: [win10]
 - Browser [firefox]

**Additional context**
also can try to type GP26 but not accepted

brentru, 2023-01-23:
@kllsamui Thanks for filing an issue. Could you please expand on "device config questionable"? what is questionable? what is not working? What do you expect to work?

Also, what is not showing data? The potentiometer connected to A0? 

Could you please include a screenshot of your WipperSnapper device page?

kllsamui, 2023-01-24:
[image]
[image]
[image]

in the device config i can choose ONLY GP0 ..3
for a POTI / Ain of a PICO_W
but there is no analog in on that ports, only at GP26 27 28
that i am NOT able to choose!

kllsamui, 2023-01-24:
@brentru 
now also do the update to 100.b59
still POTI only take GP0 show A0 and NO VALUE

[image]

brentru, 2023-01-24:
@kllsamui Thanks for the screenshots. I've assigned myself to this issue and will take a look at it.

brentru, 2023-02-03:
Possible line: https://github.com/adafruit/Wippersnapper_Boards/blob/main/boards/rpi-pico-w/definition.json#L231

brentru, 2023-02-07:
@kllsamui  Hi - this should be fixed with the release of Beta 60. I've fixed it within this pull request. https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/407

[image]

Please try out beta 60 and let me know if this is resolved.

kllsamui, 2023-02-08:
sorry, i have trouble with this..

the file 
wippersnapper.rpi_pico_w.1.0.0-beta.60.uf2
i downloaded is exactly same long as the beta-59 file

and shows up as beta-59

and no update to beta-6 option shown...

can only choose A0 GP0
but see A26 potentiometer  ! and a changing value !!!!

so yes, somehow it is working now

[image]

kllsamui, 2023-02-08:
possibly needed to wait it out??
now it shows beta 60
? possibly firefox update problem [ctrl][F5]?
also as analog value show 1023 ? 
but as voltage 0.05 ? should be 3v3 ?