adafruit/Adafruit_Wippersnapper_Arduino #408: Packet decode error, Beta60 - FeatherESP32S3psram2mb with SEN5x and 1 neopixel (onboard)

tyeth, 2023-02-13:
Similar to before, I've got a neopixel on the feather S3 2MB PSRAM, running from main branch as of an hour ago. I add the neopixel first, then on subsequent reboot I add the sen5x with 7 of the 8 metrics (not fahrenheit), and get i2c decode error with a total of 8metrics. I initially saw it sooner with the onboard lipo charger metrics active, but I culled them.
 I naively assume there was a regression related to packet size @brentru

brentru, 2023-03-02:
We could naively increase the overall buffer size, it’s currently pretty small (even compared to the size of the filesystem overhead)

brentru, 2023-03-02:
Though it might be worth it to find the size of each `sensorEvent` message, and calculate an estimated maximum for the amount of packed `sensorEvent` messages  (within message, `sensorEvents`).

brentru, 2023-08-31:
This should be resolved by https://github.com/adafruit/Adafruit_MQTT_Library/pull/227 and will be included in WipperSnapper Beta 71 (released later today).

Please try beta 71 and let me know in this issue if the packet decode error has been resolved or is still persistent.

closed 2023-08-31 by brentru