adafruit/Adafruit_Wippersnapper_Arduino #410: Dev Components don't send a component Init message

tyeth, 2023-02-13:
Devices adding 'Dev' components wont be able to use them until rebooted. This is because no Init message is sent. De-init messages are sent as expected.

Discovered while testing #409 

> Yes just this second seen another bug though, it seems to not send a Device Init message when you add a Dev component, but it works with a reboot (the component is added to the list but the device gets no component Init message).
This can now be nicely tested with the Si7021 and the SHT20 (both use same driver but one is Dev state). I believe the HU21D is also driver compatible if you have one of those and no Si7021/SHT20.

> An example of the console output for deleting the Si7021 and adding it as the SHT20 instead. It de-inits the si7021, does a scan when adding the SHT20 to verify address, then never does an Init. The same is not the case with non-dev components.
```
17:45:19.343 -> PING!
17:45:24.084 -> PING!
17:45:28.767 -> PING!
17:45:30.201 -> * NEW MESSAGE [Topic: Signal-I2C]: 
17:45:30.201 -> 4 bytes.
17:45:30.201 -> cbDecodeSignalRequestI2C
17:45:30.201 -> NEW COMMAND: I2C Device Deinit
17:45:30.201 -> Publishing Message: I2CResponse...Published!
17:45:33.492 -> PING!
17:45:38.205 -> PING!
17:45:42.908 -> PING!
17:45:45.788 -> * NEW MESSAGE [Topic: Signal-I2C]: 
17:45:45.788 -> 12 bytes.
17:45:45.788 -> cbDecodeSignalRequestI2C
17:45:45.788 -> I2C Scan Request
17:45:45.788 -> EXEC: I2C Scan
17:45:45.788 -> Found I2C Device at 0x11
17:45:45.788 -> Found I2C Device at 0x64
17:45:45.788 -> I2C Devices Found: 2
17:45:45.788 -> Publishing Message: I2CResponse...Published!
17:45:47.626 -> PING!
17:45:52.319 -> PING!
17:45:57.020 -> PING!
17:46:01.762 -> PING!
17:46:06.488 -> PING!
17:46:11.169 -> PING!
17:46:15.884 -> PING!
17:46:20.610 -> PING!
17:46:25.041 -> STATUS LED BLINK KAT
17:46:25.312 -> PING!
17:46:30.019 -> PING!

```

brentru, 2023-02-13:
@lorennorman  Could you please look into this? Seems to be on the broker side.

lorennorman, 2023-02-21:
On it!

Don't see the problem yet but I do have some I2C errors from last week that might be related. I just improved the error messages so I get more info from them, @tyeth if you wouldn't mind going and triggering the issue again, that might uncover something fast for me.

Otherwise I'll keep tracking this down this week!

tyeth, 2023-02-21:
@lorennorman I haven't exactly recreated it. I did once get it to reboot when doing an I2c de-init, which it hasn't re-done, and I stupidly wasn't running serial monitor. I kept trying adding the Si7021 and then removing and adding as the SHT20 (dev component version using same driver as si7021), while fiddling the neopixel colour between sensor polling, but never saw the same lack of De-init. I did just try deleting device and re-adding. Separately I don't always get a reading off a newly added sensor until after the period elapses, for example I set the si7021 to 15minutes this time and saw no readings, then changed to 30seconds and got one immediately, but the latest attempt hasn't sent the init message for the SHT20, my page is still waiting for results and my serial monitor just says ping endlessly:
[image]
Device link/name:
https://io.adafruit.com/tyeth/wippersnapper/feather-esp32s3-4mbflash-2mbpsram13617589

tyeth, 2023-02-21:
Sorry, bit stoned, thought I was looking for a de-init message, but it was the init message like the problem highlighted and recreated approximately 1hour ago

closed 2025-09-15 by tyeth