adafruit/Adafruit_Wippersnapper_Arduino #418: Raspberry Pi Pico W doesn't connect to WipperSnapper

Nwllen, 2023-03-06:
**Describe the bug**
Raspberry Pi Pico W doesn't connect to WipperSnapper. After the whole tutorial to connect the device in WipperSnapper, in the page 8 "Connect", Raspberry pi pico w only blinks it's builtin led on incrementing number of led flashes (starts with 3 and ends with 6 i think). Then the led stays on for a few seconds and the device disconnects from the USB and reconnects. The process keeps repeating

***Raspberry Pi board***
Raspberry Pi Pico W

**To Reproduce**
Followed the steps in the instructions and the device doesn't connect.

**Expected behavior**
Connecting the Pico to WipperSnapper

**Which components are connected to your device**
Raspberry Pi Pico W

**Screenshots**
-

**Desktop (please complete the following information):**
 - Windows 10
 - Google Chrome
 - Version 110.0.5481.178 (Official Build) (64-bit)

**Smartphone (please complete the following information):**
 - 

**Additional context**
Add any other context about the problem here.

brentru, 2023-03-30:
@Nwllen I'm unable to replicate this issue on my end:
[image]

Let's try the following:

Please check the `secrets.json` file on the WIPPER drive contains the correct Adafruit IO Username, Adafruit IO Key, WiFi SSID, and WiFi Password.

Then, could you please copy and paste the contents of `wipper_boot_out.txt`, on the Wipper drive, before the device restarts

indecorous, 2023-05-17:
Hello all,
Has anyone come up with a solution for this? I’m having the same issue with the reboot loop on a Pico W after Wippersnapper install. I brought this up in the Adafruit forum but not much is happening there either. It feels like this is just being brushed off. I really expect better from Adafruit.

Thanks for your time.