adafruit/Adafruit_Wippersnapper_Arduino #421: PMSA003I Reads 0ppm

videopixil, 2023-03-18:
**Describe the bug**
PMSA003I Reads 0ppm. CircuitPython example code reads values higher than 0. Particles .3 and .5um frequently show higher values. Particles 2.5 through 10um frequently show 0 values more often. 

***Arduino board***
Feather ESP32-S2 and Feather TFT Reverse S3

**To Reproduce**
Steps to reproduce the behavior:
add component in wippersnapper

**Expected behavior**
WipperSnapper Values should show values shown in CircuitPython code

**Which components are connected to your device**
PMSA003I

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Browser: safari
 - Version 16.3

**Smartphone (please complete the following information):**
 - Device: iPhone 13pro
 - 16.3.1
 - Browser safari
 - Version 16.3.1

**Additional context**
Add any other context about the problem here.

brentru, 2023-04-10:
Soaking this overnight

brentru, 2023-11-02:
Unable to reproduce on latest build of Wippersnapper, please update your board's firmware and try again.

closed 2023-11-02 by brentru