adafruit/Adafruit_Wippersnapper_Arduino #422: I2C fails to find device

HunterFray, 2023-03-28:
I2c scan does not find the STEMMA soil sensor I have attached to the pico w I am using, I have confirmed that pins for SCL and SDA are connected to the correct pins and that the pico is able to communicate with the sensor if circuitpython is installed instead of wippersnapper.

No matching, unused addresses found! is the error shown when trying to add the component.
Running an i2c scan from the dashboard for the pico also finds no usable addresses.

[image]

brentru, 2023-04-10:
@HunterFray  Do you know which version of WipperSnapper you're running on the PicoW?

brentru, 2023-04-10:
@HunterFray  

I'm unable to replace this on WipperSnapper Beta v61 connected to a STEMMA soil sensor. Please verify your wiring and connections and try again.

If that does not work - please attach a photo of your wiring (top-down is preferred) to this issue.

Create Soil Sensor Form:
[image]

brentru, 2023-06-30:
Due to lack of response and inability for me to reproduce it, I am closing this issue.

Please feel free to re-open it and tag me if you are still having this issue.

closed 2023-06-30 by brentru