adafruit/Adafruit_Wippersnapper_Arduino #423: Scheduled Reactive Actions

rob-vaughn, 2023-04-05:
@brentru

I'm designing a temperature control system for a growth chamber experiment and would like to use Adafruit IO to control the heating and cooling components. Reactive actions, at this time, could be used to maintain a constant temperature by turning on relays that control air conditioners and heaters, but if I'm looking to change the target temperature throughout the day based on a preset schedule, I don't think the current actions can do that unless I manually change the target temperatures.

If there was an option to set the time of day that reactive actions are initiated and for how long, I could create, for example, 12 2-hour reactive actions that occur over a 24-hour period (reactive action number 1 12:00am to 2:00am; number 2 2:00am to 4:00am; etc.).

On another note, I'm also designing an outdoor growth chamber system that will maintain temperature differentials between ambient temp and chamber temp. It would be super handy if reactive actions could be triggered by one temp sensor value being so many degrees greater than (or less than) another temp sensor. 

Anyway, just some ideas for additional features!

Thanks!

brentru, 2023-04-06:
@rob-vaughn  - @jwcooper  or @lorennorman  may know more than me on this, I'll defer to them for if it's achievable or not.

rob-vaughn, 2023-04-17:
@jwcooper <https://github.com/jwcooper> or @lorennorm
<https://github.com/lorennorman> is this something that the program could
be set up to do?

Thanks.

rob-vaughn, 2023-04-17:
> Thanks for your response, Loren!
>
> Yes, I would like to use Adafruit IO actions to control the temp.
>
> What I'm trying to do is to create, for example, a 12-step temperature
> schedule that changes the target temp in a reactive action every 2 hours
> (or as often as possible). Can a scheduled action change the target temp of
> a separate reactive action?
>
> For the cooling and heating controls, you're right that I would need
> separate reactive controls for each and a set of these separate controls
> for every time step. All I would do is have the cooling action trigger a
> relay when the temp is 1°C above my target and the heating action trigger a
> relay when the temp is 1°C below my target. Not exactly a PID controller,
> but this is basically how an Inkbird temp controller works.
>
> When you mentioned "This is tough, unless the target temperature is itself
> a feed value.", is it possible to create a repeating feed of preset values?
> If so, this may work, but it would probably be more complicated than having
> a version of the reactive action that can be scheduled. I imagine a
> scheduled reactive action looking like this (I photoshopped some pics):
>
> [image: image.png]
>
> But, I guess there would have to be an end time ("start every day at
> 12:00, End every day at 14:00")? Or would the most recent scheduled event
> be prioritized? If this were possible, I would have 24 of these (12 for the
> cooling relay and 12 for the heating relay), and this would repeat every 24
> hours.
>
> Thanks again for your help!

Update: I figured out how to make a  scheduled action that updates a feed I
call "temperature schedule". I then made a reactive action that uses the
the current value in the temperature schedule feed as the target temp. I
think this will work.

One thing I can't figure out is why the new values don't seem to be
published to the temperature schedule feed at the specified time. Might
have something to do with this statement on the action page: "Schedule
triggers poll every 5 minutes. The most often a trigger can run is every 30
minutes". I'm not quite clear on what this means.

Thanks!

lorennorman, 2023-04-18:
It sounds like you got this figured out!

> why the new values don't seem to be published to the temperature schedule feed at the specified time.

Can you screenshot one of your Scheduled Actions? And some of the data points and their timestamps that came from this Action? (your last screenshot via email didn't come through on GitHub, btw) 

One thing that comes to mind is your timezone setting, have you set that up in your account profile? If not, then it defaults to UTC, which would probably confound things.

On the "every 30 minutes" thing, if you aren't trying to execute 2 Scheduled Actions within 30 minutes of each other, they should happen when they say they will (within a couple minutes, anyway.) An IO Plus subscription drops that limit down to 5 seconds, if you need that!

rob-vaughn, 2023-04-21:
Hi Loren,

The scheduled actions would eventually go, but they would happen around 3
to 4 minutes after they were scheduled. I have the IO+ account, so the
limit shouldn't apply to me. It might have been because I literally
scheduled it to happen within minutes of me making the action. I'm not
sure. I don't have a screenshot of the time stamps because I deleted the
component. Sorry.

The time zone thing was tricky to work out because I'm currently in ADT,
but there is only AST as an option, so I had to pick some random GMT-3 time
zone to make it work. An option for daylight saving time would help this
and keep time accurate throughout the year. Just saying...

rob-vaughn, 2023-04-24:
> @rob-vaughn <https://github.com/rob-vaughn>
>
> Scheduled actions should clear within the minute you scheduled them at.
> Let us know if that isn't happening most of the time.
>
> On the timezone, are you saying you want to ignore Daylight Savings? Or
> automatically observe it?

I'll let you know if it happens again.

For time zones, I'd like the option to automatically observe daylight
savings. For Atlantic time, this would be very helpful!

Thanks.

lorennorman, 2023-04-24:
Ok quick answer:
- if you WANT magic DST handling, you should select AST (for now) to represent Atlantic Time
- if you don't want it, you select a UTC-X timezone, and you'll always be pegged to UTC

Yes, we realize it should show ADT right now, not AST, for a DST-observing Atlantic Time Zone setting. That is a separate bug that we'll address shortly. (If it's deployed before you go and change things, you'll see ADT instead of AST. If it's no, and you set AST now, it'll magically change to ADT when we deploy this.)

Aside, I'm surprised you want to observe DST for this temperature-at-time-of-day code? Isn't this an example where you'd want to be pegged to the physics of the planet vs the social clocks of the humans upon it? Or maybe it is more about a work schedule than physics? Just curious!

Thanks for all this feedback and clarification, IO gets better through this process.

rob-vaughn, 2023-04-24:
Very true--for long-term experiments, using standard time would be best.
For shorter experiments that take place in the summer during ADT, it would
be handy to have that option. Otherwise, in order to have the scheduled
actions line up with current DST, I'd have to pick some random GMT-3 time
zone or remember to subtract an hour in my head every time I schedule
actions.

Anyway, thanks for your help! Always happy to provide feedback.

closed 2023-09-06 by brentru