adafruit/Adafruit_Wippersnapper_Arduino #424: Wippersnapper/AIO does not detect Adafruit Huzzah32 V2

zeropotentialenergy, 2023-04-06:
Adafruit Huzzah32 V2 (ESP32 chip, 8MB Flash, 2MB PSRAM, external antenna connector) when flashed with Wippersnapper does not connect to wireless network.  As a result, the board is not detected by Adafruit IO.  The status LED is solid yellow for several seconds, then off, then on again for several seconds.  I suspect a boot loop.  I have two identical Huzzah32 V2 boards.  Both have successfully connected to the wireless network when flashed with CircruitPython.

[image]

[image]

**Expected behavior**
Connect to wireless network and AIO

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?
None currently

**Desktop (please complete the following information):**
ASUS Win 10 laptop
Chrome 111.05563.147
Nothing else plugged into laptop

I'm an amateur so feel free to ask questions if I left out some important details.  Thank you in advance.

jessecheu, 2023-04-08:
Hi, I am having the same issue. When I connect it to an IDE, I see the error message
 ledc: ledc_set_duty(647): LEDC is not initialized
ledc_update_duty(647): LEDC is not initialized.
It seems like there's a doom loop.

jessecheu, 2023-04-11:
@brentru 
Yeah, I ended up getting it to work. I had to connect to a 2.4GHz network though, I was kind of expecting a popup after the installation but it seems like the popup came when I went to URL https://io.adafruit.com/wippersnapper

brentru, 2023-09-06:
@zeropotentialenergy Hi - we've upgraded the site used to flash WipperSnapper to devices. Please try this again with https://io.adafruit.com/devices/new/feather-esp32-v2/web/step-2 and let us know how it turns out! 

I am closing this issue, if you are still experiencing problems with the new web flasher - please comment and I'll re-open it.

closed 2023-09-06 by brentru