adafruit/Adafruit_Wippersnapper_Arduino #425: Adafruit ESP32 Feather Huzzah not supported?

mik00, 2023-04-10:
PlatformIO.ini does not mention the **Adafruit ESP32 Feather Huzzah** (i.e.  good old v1!)  

Is there some general issue getting it to work under PlatformIO, or do I use some other environment ID?
I see the ESP32_V2 and the ESP8266-ADAFRUIT_HUZZAH

Thx

brentru, 2023-04-10:
@mik00  This Feather is supported by WipperSnapper, a direct link for installation via our web tool is here: https://adafruit.github.io/WipperSnapper_Firmware_Uploader/?board=feather-esp32

Are you intending to locally compile WipperSnapper for development on PlatformIO, or just to get it running on your feather?

mik00, 2023-04-12:
@brentru Thanks for that pointer. It was a long, and embarassing struggle to track it down.  Knowing I had your working project was very helpful in eliminating all the red-herrings.  My faithful old ESP32a work all work just fine now.  

The problem was that an earlier program, had not been erased.  It was occasionally flashing the built-in LED for 50ms, a LED which, as we all know of course, resides on GPIO #13. Duh!  It happened just often enough that it would prevent a successful re-flash.

Thank you!
//mik00

brentru, 2023-04-12:
No problem, feel free to open a new issue if you experience further problems!