adafruit/Adafruit_Wippersnapper_Arduino #426: Wipper Snapper on Arduino MKR Wifi 1010

evil-electronics-klub, 2023-04-12:
**Describe the bug**
In file included from /Users/joevella/Documents/Arduino/libraries/Adafruit_WipperSnapper/src/Wippersnapper.h:47:0,
                 from /Users/joevella/Documents/Arduino/libraries/Adafruit_WipperSnapper/src/network_interfaces/Wippersnapper_WIFININA.h:25,
                 from /Users/joevella/Documents/Arduino/libraries/Adafruit_WipperSnapper/src/Wippersnapper_Networking.h:37,
                 from /private/var/folders/62/fxgr2sqs0gl0gnbvw4wmz8j00000gn/T/.arduinoIDE-unsaved2023312-18123-32cm6n.q9g4e/Wippersnapper_NoFS/Wippersnapper_NoFS.ino:19:
/Users/joevella/Documents/Arduino/libraries/Adafruit_WipperSnapper/src/components/ds18x20/ws_ds18x20.h:22:10: fatal error: DallasTemperature.h: No such file or directory
 #include <DallasTemperature.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

***Arduino board***
Arduino MKR Wifi 1010

**To Reproduce**
Steps to reproduce the behavior:
Compiling the Nofs example code

**Expected behavior**
expected no errors after entering wifi and IO info

**Which components are connected to your device**
none

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
Mac OS 10.13

brentru, 2023-04-12:
@evil-electronics-klub  Could try installing the https://github.com/milesburton/Arduino-Temperature-Control-Library via the Arduino Library Manager and re-compiling?

evil-electronics-klub, 2023-04-12:
I'll try that, but what's weird is this is a new board. The only thing previously setup was a basic WifiNINA sketch to get wifi connectivity. Just looking to get it connected to Adafruit IO and start connecting things to it

brentru, 2023-04-12:
@evil-electronics-klub  This is a bug, the library should have installed it "for you". Please let me know if this works for you, adding tag "awaiting feedback"

evil-electronics-klub, 2023-04-13:
Installed the library (along with another called OneWire that seemed to be needed). The sketch was then able to compile and load. However no matter what I try, the board doesn't seem to want to connect to wifi. Just an endless series of "trying to connect..." and reloads :(

brentru, 2023-04-13:
@evil-electronics-klub  Are you on a 2.4GHz Wireless network? Can you confirm that the WiFi credentials within the `secrets.json` file are correct?