adafruit/Adafruit_Wippersnapper_Arduino #427: Change Credentials without firmware upgrade

DJDevon3, 2023-04-18:
There's no way to easily change Wifi credentials other than with a firmware upgrade. Recently had to reset my wifi password and change everything.  WipperSnapper gave me more issues than any other IoT device I had to reset.

***Arduino board***
Adafruit Feather ESP32-V2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to AdafruitIO
2. Click on Device
3. Scroll down to Change Credentials
4. See error, it doesn't exist.

**Expected behavior**
A simple link to change the credentials on the WipperSnapper device.

brentru, 2023-04-24:
@lorennorman  We may want to address this in the web upload refresh

brentru, 2023-11-02:
@lorennorman  From the new web uploader, can a user now update the credentials of a device without re-installing the entire WipperSnapper binary?

lorennorman, 2023-11-02:
@brentru I thought we did support that but I'm not seeing it just now. Making a note to investigate soon, might be an easy add.