adafruit/Adafruit_Wippersnapper_Arduino #428: ESP32-V2 Webflasher Connects then locks

DJDevon3, 2023-04-18:
**Describe the bug**
[Adafruit Discord Discussion](https://discord.com/channels/327254708534116352/355063529302392832/1096608757561171978)
 I tried https://adafruit.github.io/WipperSnapper_Firmware_Uploader but it cuts me off immediately after a reboot.

```py
Wippersnapper Firmware Uploader loaded.
Connecting...
Connected successfully.
Try hard reset.
Error: Oops, we lost connection to your board before completing the install. Please check your USB connection and click Connect again. Refresh the browser if it becomes unresponsive.
```
WipperSnapper connects to the board and locks it before Webflasher has time to begin the process of erasing flash... would be my guess.  Was able to erase flash manually with python esptool first try.  The latency from the internet between bootup and reflashing sequence is my guess. 

I have no problem locally getting under that window after bootup, up arrow in cmd prompt, erase flash.  There is a small window where you can erase it.  I only know this from fighting with it so many times in the past.  If you time the entry in esptool command prompt just right you can beat WipperSnapper to erase the flash before it locks you out.  

The latency from the webflasher makes that time window... almost impossible.. a few seconds, 3 seconds tops with my internet speeds to where ever web flasher is running from. It isn't enough to start the flash erase procedure apparently.

***Arduino board***
Feather ESP32-V2

**To Reproduce**
Steps to reproduce the behavior:
1. Have an ESP32-V2 with latest updated WipperSnapper
2. Click on webflasher url listed above
3. Connect to board (paired) and attempt to flash
4. It will connect successfully and 1-3 seconds later disconnect.

**Expected behavior**
Update firmware using webflasher

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome

**Additional context**
I might not remember all of the steps that happened in exact order. It was 4 days ago and my memory is horrible.

brentru, 2023-09-06:
@DJDevon3 I am unable to reproduce this issue with the latest version of io.adafruit.com and wippersnapper. We've updated/upgraded the web flasher as well, it now runs from io.adafruit.com instead of elsewhere. Please try that out and feel free to re-open this issue if you are still having trouble.

closed 2023-09-06 by brentru

DJDevon3, 2023-09-07:
The updated one with adafruitio works much better. Yep can close this one. 👍